SolidCode/SolidPython #25: Consider making code PEP8 compliant

sorki, 2015-01-12:
There's autopep8 tool that can do this automatically.

Would make it easier to contribute to this project.

etjones, 2015-01-12:
Thanks.  I may or may not get to this at some point.  I ran autopep8 on the solidpython module and the changes it recommended decreased readability for me in some of the tabular data at the beginning of the file.  I like that PEP8 exists as a general guideline, but especially in places where there's a lot of parallelism in code (dictionaries with lots of identical fields, sequential method calls with parallel arguments) I really like having extra whitespace to emphasize what's going on, which is in respectful disagreement with PEP8's Pet Peeve section.  So... there are definitely some things I could clean up, but I'm not in a big hurry to do it, and I'm pretty likely to never move to 100% orthodox PEP8.  If there's some formatting that really galls you, feel free to change it and send a pull request; I'd be happy to take a loog.

etjones, 2015-05-23:
Hey, just BTW, I ran autopep8 on everything in SP today.  I went back and made plenty of changes -- there's lots of indented, tabular data where I kept formatting, and I kept 80+ char lines where a longer line felt more readable than when it was broken into two lines.  Not totally kosher, but better than it was.  Thanks for giving me the nudge those months ago; it's an improvement.

closed 2015-05-23 by etjones