SolidCode/SolidPython #28: Keep alive an openscad instance to access its caches for better performance

goatchurchprime, 2015-03-02:
My experiments with solidpython in my (quite rough but very handy) development environment suggest that performance is going to be a big issue.

The development iteration function (which is too slow on complex models) is as follows:

https://bitbucket.org/goatchurch/twistcodewiki/src/91764d2feb87dcfe97f86d31c6dec9705943ba90/sparepy/solidpythonexpts.py?at=default

```python
retriang = "\s+vertex\s+(\S+)\s+(\S+)\s+(\S+)"*3
tfile = tempfile.NamedTemporaryFile(suffix='.scad')
tfile.write(scad_render(d).encode('utf-8'))  # scad_render_to_file() not working
tfile.flush()
stlfile = tempfile.NamedTemporaryFile(mode='r', suffix='.stl')
retcode = subprocess.call(["openscad", "-o", stlfile.name, tfile.name])
stlverts = re.findall(retriang, stlfile.read())
triangles = [ tuple(map(float, vs))  for vs in stlverts ]  # 9-tuples
```

Any thoughts on keeping alive an OpenSCAD instance in such a way that I can pipe in new versions of the scad text and rerunning it as though it's been types so that it can use its internall caches?

spresse1, 2015-03-02:
OpenSCAD has the "Design -> Automatic Reload and Compile" option for 
this purpose.  It will watch for changes in the file and rerun it on 
changes.

Steve

etjones, 2015-03-03:
I'd second the "Automatic Reload and Compile" option above; that's how I've always done my development.  OpenSCAD's preview mode is slow for large models, its more accurate CGAL rendering mode is slower, and the STL generation is slowest of all.  The SolidPython portion of that process doesn't have much effect on the time; SP is basically just a string-manipulation system to translate Python code into valid OpenSCAD code and all the actual hard work is in the OpenSCAD kernel.

On another note, if scad_render_to_file() isn't working for you, that IS a bug I'd love to hear about and fix.  If you've got a small repeatable example, I'd be grateful if you'd make a new issue for it.  Cheers!

closed 2015-03-03 by etjones