SolidCode/SolidPython #29: SolidPython missing support for recent OpenSCAD statements in 2014.03

etjones, 2015-03-03:
Looking at the [OpenSCAD cheatsheet](http://www.openscad.org/cheatsheet/index.html), there are a few statements ('text', 'offset') not supported in SolidPython.  Putting this here as a placeholder to remind myself to add these pieces

ehooo, 2015-04-01:
Also '[concat](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#concat)', '[let](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#led)' and '[assign](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Conditional_and_Iterator_Functions#Assign_Statement)'.

etjones, 2015-05-09:
From the OpenSCAD 3/2015 [Release Notes](https://github.com/openscad/openscad/blob/master/RELEASE_NOTES):

**Language Features:**
* Added text() module for 2D text
* Added offset() module for 2D offsets
* Added list comprehensions and let()
* Added concat() function
* Added chr() function
* surface() can now take PNG images as input
* min() and max() can now take a vector argument
* 2D minkowski can now handle polygons with holes
* Variables can now be assigned in local blocks without using assign()

etjones, 2017-01-02:
All the features mentioned here have either been added (text(), offset()) or are machinery in OpenSCAD that's already available in Python (let(), assign()).  Call it good for now

closed 2017-01-02 by etjones