SolidCode/SolidPython #33: hole operator is not found

jdawgaz, 2015-04-08:
did the pip install last night for python 2.7.8
wrote a program example, all looked good.
UNTIL I tried the hole_example.

hole() is not found, contrary to the documentation.

Traceback (most recent call last):
  File "hole_example.py", line 86, in <module>
    a = pipe_intersection_no_hole() + right( 45)(pipe_intersection_hole())
  File "hole_example.py", line 26, in pipe_intersection_hole
    pipe_a = outer + hole()(inner)
NameError: global name 'hole' is not defined

Nesciosquid, 2015-05-05:
Have you installed the latest version of OpenSCAD? I was a little ways behind and it was screwing a bunch of things up.

etjones, 2015-05-09:
Thanks for pointing this out.  The problem is that the version on PyPi is about a year behind what's in Master on Github, and the old SolidPython version (installed by Pip, without `hole`)  was being used on the newer hole example.  I'll look to push the newest version (holes included) to PyPi in the next couple weeks.  

In the meantime, here's a potential workaround:
``` shell
pip uninstall solidpython
cd $YOUR_PROJECT_DIR
git clone https://github.com/SolidCode/SolidPython.git
cd SolidPython
# may need sudo, depending on your setup
python setup.py install
# Now run examples; hole should work
cd solid/examples/
python hole_example.py
```

closed 2015-05-09 by etjones