SolidCode/SolidPython #34: use() is mangling Windows paths

Nesciosquid, 2015-05-05:
I'm trying to employ the use() function to grab SCAD files out of a subdirectory of the folder I'm running my SolidPython scripts from.

Regardless of whether I use relative or absolute paths, using the `os.path` functions to specify the target file results in a ValueError from `_get_include_path()`.

The only way I was able to get this to work was by avoiding `os.path` altogether and specifying relative paths as strings using forward slashes, or creating an absolute path with `os.path.abspath` and using `string.replace("\\", "/")`, both of which are pretty gross.

This only happens with folder or file names beginning in letters that correspond to backslash ASCII codes -- having a folder named "features" breaks it, but "deatures" works just fine. This is definitely some kind of backslash escape problem, similar to http://stackoverflow.com/questions/11815881/windows-filename-gets-mangled, since it only happens with folder names beginning with a particular character (in this case, a folder named "features" doesn't work, but "deatures" does.) 

Am I doing something stupid, here, or is there some kind of wonky behavior going on? 

I've also included a screenshot of what the OpenSCAD file looks like if I modify `_get_include_path` to return `os.path.abspath(include_file_path)` instead -- maybe there's something weird happening when writing the include/use line into the actual SCAD document?

[image]
[image]
[image]
[image]

etjones, 2015-05-09:
Hey, sorry not to get to this sooner.  I spun up a Windows VM to check this out and couldn't duplicate the problems you had right away.  However, I don't think I've used SolidPython in Windows for more than a year, and I'm not at all certain there's no problem.  Could you upload the code you were trying to run?  

You might check out `SolidPython/solid/examples/basic_scad_include.py`.  I had no problems running that one either with Git Bash or the normal Windows command prompt.  Let me know if that works for you and that might help track things down.

I notice in the last screenshot you attach that the directory you're using contains a non-Ascii ankh character ("SCAD<ankh>eatures") and it looks like that gets translated to "SCAD<ff>eatures" in the generated SCAD code.  Might that be some kind of issue?

Let me know how things go, and thanks for checking out SolidPython.

Nesciosquid, 2015-05-10:
Here's a .rar archive of some example code that shows the difference in behavior when you specify the path in different ways.

That non-ASCII ankh is the result of the mangling, not the cause. :)

https://www.dropbox.com/s/g1i8yk6s8qm5hte/solidpython_issue_34_example.rar?dl=0

etjones, 2015-05-10:
Found it.  SolidPython generates a Python code string to wrap around any imported SCAD code, and it evals that generated string.  The backslash-delimited Windows paths were getting interpolated with only a single backslash character, leading to the problems you saw.   I escaped the backslashes one more level before the eval and everything seems to be working smoothly now.  Thanks for bringing that one up.

closed 2015-05-10 by etjones

Nesciosquid, 2015-05-10:
Awesome!

Thanks for getting to this so quickly. 

I'm making some tools for my PhD research that use your library, and it's awesome to know you care about maintaining it!

-Aaron Heuckroth

etjones, 2015-05-11:
Glad to know SolidPython is helping you get stuff done!  Best of luck,

Evan