SolidCode/SolidPython #38: Converting openscad_builtins to real types

steveb, 2015-08-09:
This would allow autocompletion in code editors, and eventually full sphinx documentation of the arguments to builtins.

I'd be happy to put together a pull request to implement this if you're receptive.

etjones, 2015-08-10:
Hi Steve- 
   That sounds like it could be a useful addition!  All the classes in openscad_builtins are now built from that big list of dictionaries at the top of solidpython.py. Am I right to think you want to exchange those ~30 dictionaries for explicit classes (e.g.  ```class square(OpenSCADObject):
        def __init__(self, size, center=True):
            OpenSCADObject.__init__(self, 'square', {'size':size, 'center': center})
```)  so that they can be pre-analyzed by code editors?  If that's the case, that sounds fine. I suspect that having all those classes explicitly present rather than created by an opaque code generator would be helpful for a lot of people.  Let me know what you come up with, and I'd be happy to take a look.

steveb, 2015-08-10:
As a starter I'm just going to print class_str before the exec and paste the resulting classes as source. It seems to work just fine.

etjones, 2015-08-10:
That’s exactly how I’d do it. Thanks!

closed 2015-08-12 by etjones

mfeif, 2015-11-01:
As long as Steve is taking a crack at this, might I make a request/suggestion?

A weirdness in the syntax is not necessary (unless I'm mistaken):

```python
d = difference()(
    cube(10),  # Note the comma between each element!
    sphere(15)
)
```
A more "pythonic" form would simply be:
```python
d = difference(
    cube(10),  # Note the comma between each element!
    sphere(15)
)
```
(The difference is no paren-paren right after difference.)

Similarly, `translate([0, 10, 0])( cube(5) )` can just become `translate([0, 10, 0], cube(5) )`

Right?

Seems like the extra instantiation of the proxy is there just to make the auto-generation of proxy classes possible. If we cook that once into actual source (instead of a string that is exec'd), then edit by hand as necessary, that might be nicer.

Various other function protocols can be explored, like sending the children in as an iterable or whatever, rather than inline args...

Thoughts?