SolidCode/SolidPython #41: scad_render_to_file() crashes when solidpython is used in a shell

mfeif, 2015-10-30:
Hi there. Just wanted to try out SolidPython, and was about to try hacking around with it from within ipython. I got the following crash when trying to make some elementary geometry:

```python
AttributeError                            Traceback (most recent call last)
<ipython-input-15-dd232dee5742> in <module>()
----> 1 scad_render_to_file(d, f)

/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/solid/solidpython.pyc in scad_render_to_file(scad_object, filepath, file_header, include_orig_code)
    143     rendered_string = scad_render( scad_object, file_header)
    144 
--> 145     calling_file = os.path.abspath( calling_module().__file__)
    146 
    147     if include_orig_code:

AttributeError: 'NoneType' object has no attribute '__file__'
```

It seems reasonable to me that people might try and use this in a REPL like ipython or even just python console.

A test for `sys.__stdin__.isatty()` would be sufficient to flag this and amend your constructed comment.

I could probably make a pull request if this is agreeable to you.

etjones, 2015-10-30:
Thanks for the report.  This was fixed in #18, (and works as expected in Master right now) but hasn't made it into a PyPI release.  I held off on that for a while waiting on a dependency to get Python3 support, but I've let it go too long.  I'll try to get a new PyPI release soon, and until then cloning from Master should solve your issue.

etjones, 2015-10-30:
Duplicate of issue #18

closed 2015-10-30 by etjones

mfeif, 2015-10-31:
Sorry, I looked in open issues for duplicates, but not closed ones :-)

Thanks!

etjones, 2015-10-31:
No worries-- it points out how much I need to get a new release to PyPI!