SolidCode/SolidPython #42: Parametric models: dealing with openSCAD variables ?

bertrand-caron, 2015-12-27:
Hi,

An interesting feature of `openSCAD` is writing parametric objects with explicit named variables.

```
a = 4.0;
cube([a, a, a]);
```

Obviously the following Python code loses this information:

```
from solid import *

a = 4.0
d = cube([a, a, a])
print(scad_render(d))
```

Have you thought about implementing this feature somehow ? 

I guess you would need to construct / pass along an 'environment' for your object, and implement some sort of substitution mechanism / formatting syntax. Maybe something like :

```
from solid import *

a = 4.0
d = cube([('{a}', '{a}', '{a}'])
print(scad_render(d, a=4.0))
```

etjones, 2016-01-05:
Hi Bertrand - 
    You're completely right about the usefulness of named parameters-- it's what made me want to use OpenSCAD in the first place! I've always used the SolidPython source as the code I wanted to work with, though, and treated the generated SCAD code as a semi-opaque layer that just gets fed to the visualizer.    To use your example, I tend to stick with something like:

```
from solid import *

a = 4.0
d = cube([a, a, a])
scad_render_to_file(d, '/path/to/your_file.scad') 
```
I then open an OpenSCAD window, open the file at `/path/to/your_file.scad`, and check the Design => Automatic Reload & Compile menu option.  Then any time I run the SolidPython code, the OpenSCAD window updates automatically.  

There are definitely times when a closer mapping between the SolidPython & generated OpenSCAD would be desirable.  However, it would mean a pretty fundamental reorganization of the code, and I'm not sure how useful this would ultimately be.  For example, this SolidPython code:

```
from solid import *

cubes = []
for i in range(1000):
    c = translate([2*i, 0, 0])(
        cube(1)
    )
    cubes.append(c)
    
scad_render_to_file(union()(cubes))
```

translates to about 3000 lines of OpenSCAD, as opposed to the much more reasonable SCAD translation:
```
for (i=[0:1000])
{
    translate([2*i, 0, 0]){
        cube(1);
    }
}
```
But... those are the perils of machine translation of any kind.  I've found over time that I just stopped paying attention to what the SCAD code looked like at all; the Python was all I needed.  YMMV.

closed 2016-01-05 by etjones

arpruss, 2016-11-05:
A useful feature of parametric output would be that one could upload to thingiverse and other users could use their Customizer app.