SolidCode/SolidPython #44: use <normal_quartz.scad>

amunizp, 2016-02-28:
How can I use the use command in the python code?

amunizp, 2016-02-28:
Example code I would like to replicate
`
/*
model of a group of quartz needles

see The Quartz Page
http://www.quartzpage.de/index.html

especially the
Introduction under Crystals
also see
Habits under Crystals

This model shows the Normal Habit
also called the Prismatic Habit
or the Maderan habit

it also has a single needle showing
the trigonal habit.

pcm  May 2015

*/

 use <normal_quartz.scad>
 use <trigonal_quartz.scad>

  scale(0.5)
  difference()
  {
  union()
  {
  sphere(r= 60, center=true);
  rotate([23,0,0])
  translate([0,0,90])
  normal_quartz(1.5767,50);
  rotate([0,25,0])
  translate([0,0,90])
  normal_quartz(1.5767,55);

 }
  translate([0,0,-62])
  cube(125,center=true);
}
`

amunizp, 2016-02-28:
Found it right at the top of the manual. Sorry.

*Face palm*

Thanks!

closed 2016-02-28 by amunizp

etjones, 2016-03-04:
Glad you worked it out!