SolidCode/SolidPython #45: ImportError: No module named 'solidpython'

vishakhkumar, 2016-02-28:
File "//anaconda/lib/python3.4/site-packages/solid/__init__.py", line 4, in <module>
    from solidpython import *
ImportError: No module named 'solidpython'

No idea why this is happening - currently using python 3.x on OS X if that helps. I've reinstalled the module which leads to the message 

> pip install solidpython

Requirement already satisfied (use --upgrade to upgrade): solidpython in /anaconda/lib/python3.4/site-packages
Requirement already satisfied (use --upgrade to upgrade): euclid in /anaconda/lib/python3.4/site-packages (from solidpython)

Upgrading it says everything is up to date. Probably because I just installed it.

vishakhkumar, 2016-02-28:
Installed it manually using the setup file. I guess that works!

closed 2016-02-28 by vishakhkumar

etjones, 2016-03-04:
Glad you worked this out! The master branch has had Python 3 support for a while now, but some dependencies weren't working with Py3 so I was hoping they'd sync up before pushing a new release to PyPI so that a pip install would work with Py3.  Thanks for the reminder and I'll try to get to this sooner than later.

voglster, 2016-12-01:
Just a heads up, this is still an issue today, same error anaconda3 python 3.5.2, ill go the setup.py path myself as well