SolidCode/SolidPython #53: Install command in readme doesn't seem to work

daveansell, 2016-10-09:
in the readme the install command is:
sudo python setup.py --install
I found I had to use:
sudo python setup.py install

thanks
   Dave

arpit1997, 2016-10-30:
@daveansell 
Thank you for reporting the bug. I guess this might just be a typo. because this is a very old version thing.
and perhaps ..
Glad to hear that you figured it by yourself.But you might consider changing the tile from command not work to `typo in the command`. It might be much helpful becuase this is no bug just a silly mistake,
Hope this is reasonable.
Thank you :smiley_cat: :smiley_cat:

etjones, 2017-01-02:
Fixed in 4849d9b25b768a39bf5d6007b9d76f979d8c4178.  Will be pushed to PyPI shortly

closed 2017-01-02 by etjones