SolidCode/SolidPython #67: resize auto=true doesn't seem to work

nerdfever, 2017-02-22:
When I try to use the auto=true parameter on the OpenSCAD `resize` function, I get:

`NameError: name 'true' is not defined`

Python 3.4 output:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\WinPython-64bit-3.4.4.2\python-3.4.4.amd64\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 841, in runfile
    execfile(filename, namespace)
  File "C:\WinPython-64bit-3.4.4.2\python-3.4.4.amd64\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 103, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)
  File "U:/Users/Dave/data/Projects/3D modeling/SolidPython/LensCap.py", line 120, in <module>
    a = LensCap(Diameter, InsideLen, OutsideLen, gap=GapWidth, clearance=Clearance)
  File "U:/Users/Dave/data/Projects/3D modeling/SolidPython/LensCap.py", line 97, in LensCap
    label = resize([innerRadius * 2 * insideTextRatio, 0, textDepth*more], auto=true)(label)
NameError: name 'true' is not defined
```
I _love_ SolidPython - many thanks to everyone who worked on it!

evandrone, 2017-02-22:
`true` is not defined in Python, but `True` is.  Bites me all the time when I go back and forth with other languages

etjones, 2017-02-22:
`true` is not defined in Python, but `True` is. Bites me all the time when I go back and forth with other languages

nerdfever, 2017-02-22:
Unfortunately that (True vs true) doesn't seem to help:

```
Traceback (most recent call last):

  File "<ipython-input-1-c1bd943c8e89>", line 1, in <module>
    runfile('U:/Users/Dave/data/Projects/3D modeling/SolidPython/LensCap.py', wdir='U:/Users/Dave/data/Projects/3D modeling/SolidPython')

  File "C:\WinPython-64bit-3.4.4.2\python-3.4.4.amd64\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 841, in runfile
    execfile(filename, namespace)

  File "C:\WinPython-64bit-3.4.4.2\python-3.4.4.amd64\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 103, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "U:/Users/Dave/data/Projects/3D modeling/SolidPython/LensCap.py", line 125, in <module>
    a = LensCap(Diameter, InsideLen, OutsideLen, gap=GapWidth, clearance=Clearance, digits=Digits)

  File "U:/Users/Dave/data/Projects/3D modeling/SolidPython/LensCap.py", line 102, in LensCap
    label = resize([newX, 0, textDepth*more], auto=True)(label)

TypeError: __init__() got an unexpected keyword argument 'auto'
```

etjones, 2017-02-22:
You're totally right, this is a bug. Sorry I was too quick on the close button.   Fix forthcoming

etjones, 2017-02-22:
Closed in 653219b5f4bb966cbf4126ce3615b358e391dfe8

Dave, if you're running the PyPI release this won't solve your problem, but if you clone the repo, `pip uninstall solidpython` and `pip install -e <$PATH_TO/SolidPython>` that should fix things up.  Thanks for catching this.

closed 2017-02-22 by etjones

nerdfever, 2017-02-23:
That fixed it.

I saw you changed object.py; I tried replacing just that file in site-packages/solid, but it didn't help. (Not sure why.)

But this worked:

`python -m pip uninstall solidpython` 
`python -m pip install -e <$PATH_TO/SolidPython>`. 

Thanks!!

nerdfever, 2017-03-01:
Related issue (unsupported parameter):

```
#    result = color([0,0,0], a=0.5)(result)
    result = color([0,0,0, 0.5])(result)
#    result = color("black", 0.5)(result)
```

Only the uncommented form works. The other two commented-out forms are supposed to work as well, per the OpenSCAD docs. But both generate complaints about unsupported/extra parameters.