SolidCode/SolidPython #69: Add to documentation

nerdfever, 2017-02-27:
Things that should be added to the documentation:

* How to use .add_param() to add special variables - for example `sphere(20).add_param('$fn', 10)`

* How to use .set_modifier() for debugging - for example `.set_modifier("#")`

* How to pass an alpha value to `color() `(I still don't know how to do it).

(Altho personally I'd find the syntax `object.highlight()`, `object.showOnly()`, `object.background(`), `object.disable()` more convenient and obvious...)

I'm not sure if there's a way for me to edit the docs on GitHub directly (if there is, I didn't find it); I'd take a crack at adding these myself.

evandrone, 2017-02-27:
All sounds good.  The docs were generated in a big [pull request](https://github.com/SolidCode/SolidPython/pull/40/files) by some very helpful guy a few months back, and tbh I'm not sure how to regenerate them.  I spent a couple hours with tox trying to work that out a couple months ago but then I saw something shiny and ran off to look at that.  Would welcome any extra knowledge about that. 

Definitely open to adding some more pythonic syntax for those aspects, although I try to balance that with retaining some similarity to the underlying OpenSCAD structure.  If you get anyplace with these issues I'd be grateful, otherwise I'll address it some Sunday.  Cheers!

nerdfever, 2017-02-27:
There's no harm in enabling it both ways - that is, allow both `.set_modifier('#')` and `.highlight()`.

That was already done with up(), down(), etc. vs. translate().

Someday if I have time (ha ha) I may pull the documentation, convert it to plain HTML, and add that stuff (and other things) myself.

SolidPython is too useful to not be fully documented! :-)

danpeirce, 2020-10-18:
Is there any documentation on:

- How to pass an alpha value to color() (I still don't know how to do it).

It is Oct 2020 now.

etjones, 2020-10-18:
Well, pull requests are always welcome...

danpeirce, 2020-10-18:
Okay I have made a pull request.
A way to proceed became more clear to me after looking closely in utils.py.