SolidCode/SolidPython #78: Automate calls to openscad

aarchiba, 2017-05-30:
Currently SolidPython stops at generating OpenSCAD files. It would be handy to have a couple of functions to run OpenSCAD as appropriate, for example

- Generate an STL directly
- Open a GUI window, possibly with a specified camera view
- Generate a PNG directly,  possibly with a specified camera view (this could allow embedding output into IPython cells, possibly including interactive widgetry)

These are all (by design) doable with subprocess/mkdtemp, but they're things one might often want out of a script generating OpenSCAD code. The main headache is conveniently dealing with people whose OpenSCAD installations are in some nonstandard place.

etjones, 2017-06-09:
I think all those things would be useful, Anne!

I toyed around with a couple options back in the day.  My thoughts then were:

a) Generating an STL in OpenSCAD takes so long for most designs that I always wanted to be able to look at it in OpenSCAD before committing to the STL render. Not to say that a direct-to-stl command isn't possible or desirable, but I suspect that if you've got things dialed-in enough to want to render directly to STL, you're probably dialed-in enough to add in one extra shell command. 

b) As you say, the difficulty in writing these extra shell scripts is knowing where to look for the executable.  On MacOS, it's pretty safe to say you're looking in `/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD`, but all bets are off on any of the many Windows/Linux variants. 

So, I'm not opposed to adding this bit of sugar, but I'm not certain how we'd go about it in a platform-independent manner.  Any ideas?

nerdfever, 2017-11-01:
The default install location on Windows (Vista, 7, 8, and 10) is:

`%ProgramFiles%\OpenSCAD\openscad.exe`

That uses the standard Windows system environment variable ProgramFiles. Most commonly it's contents are "C:\Program Files", but it's best not to count on that and just use the environment variable.

What I'd do is, while installing SolidPython, assume that's where OpenSCAD is (check to see). If it's there, fine - install. If not, stop and prompt the user for the path to OpenSCAD.

Or, if you want to be more aggressive, simply download the latest OpenSCAN an install it, if it's not already there.

closed 2018-07-23 by etjones