SolidCode/SolidPython #79: Regression from PR #65:  runaway recursion

etjones, 2017-06-09:
@conorpp supplies this code, which recurses without exit after the merged operations in PR #65:

```python
r = cube(10)
p = cube(4)
p = r * p
r -= p
scad_render(r)
```

Fix.

etjones, 2017-06-09:
Can't duplicate.  Will close before long unless I can figure out why this happened.

etjones, 2017-06-10:
This is supposed to dupe: 
```python
    r = cube(10)

    r -= cube(1)  # missed step

    p = cube(10)
    p = r * p

    r -= p

    scad_render(r)
```
Checking

hstarmans, 2017-06-23:
confirmed the current code cannot be used, get errors everywhere;
 revert to cd2574a3b98267abef290673d4272b9d0a1c39f1

git checkout cd2574a3b98267abef290673d4272b9d0a1c39f1

nickc92, 2019-02-12:
Ok, this has been fixed. The root cause was essentially the same cause as an issue of runaway recursion that I had seen, which was introduced when a change was made that was intended to streamline the generated OpenSCAD code.  The fixed version is there in git and PyPI.

closed 2019-02-12 by nickc92