SolidCode/SolidPython #86: import(...) support, dxf_linear_extrude(...) deprecated

tom-f-oconnell, 2017-12-12:
Calls to `dxf_linear_extrude` will produce SCAD code that yields the following warnings:
```
DEPRECATED: The dxf_linear_extrude() module will be removed in future releases. Use linear_extrude() instead. 
DEPRECATED: Support for reading files in linear_extrude will be removed in future releases. Use a child import() instead.
```
It seems that `import` is not implemented in SolidPython now, and it seems like the current convention of
```
from solid import *
```
might be problematic if we want to just name it import(...).  Would you just want to implement it under a different name?

It might be nice to transition to / advocate for a convention more like
```
import solid as sp
```
, but I could see you not wanting to do that for momentum reasons.

etjones, 2017-12-12:
Thanks, Tom.  That's a really useful report.  I've been OK with the `from solid import *` idiom since, unlike other Python code where you really don't want to clutter the namespace, most SolidPython programs are relatively small and single-purpose.  

I'm leaning towards `import_` for the DXF import function.  Reasonable?

There's a certain amount of work getting this fixed, but I don't think it should be much.  And I'm on vacation, so... maybe sooner than later.

etjones, 2018-07-23:
So, I think things are good as they are.  This code works in the current Master.  It prints a deprecation warning when OpenSCAD wants to let you know about a deprecation, and also shows the SCAD-approved way to work

```python
from solid import *
PATH = '/PATH/TO/square.dxf'
HEIGHT = 5

# Deprecated, and will raise a warning. But that seems like correct behavior
# until OpenSCAD actually makes dxf_linear_extrude() illegal.
a = dxf_linear_extrude(file=PATH, height=HEIGHT)
scad_render_to_file(a)

# The future-proof way:
a = import_(file=PATH)  # or: a = import_dxf(file=PATH)
b = linear_extrude(height=HEIGHT)(a)

```

closed 2018-07-23 by etjones