SolidCode/SolidPython #93: Extend screw_thread.thread to handle multiple starts and default threads with flat tops

nerdfever, 2018-07-25:
Many standard (common) thread profiles have flat tops. For example, see https://en.wikipedia.org/wiki/Unified_Thread_Standard

Here's an implementation that does both multiple starts and default thread profiles with either pointy or flat tops:

`def thread(diameter, length, lead, tooth=None, 
           toothBite=None, toothBottom=None, toothTop=None, 
           external=True, starts=1, neckInDegrees=0, neckOutDegrees=0, segments=64):
    ''' 
    Makes an internal or external screw thread.

    diameter - of the base cylinder to which threads are referenced
    length - Z axis length of the thread (goes bottom to top)
    lead - Z distance the thread moves per turn (if starts==1, this is the pitch)
    tooth - the tooth profile (see below). if None, the default profile is used.
    toothBite - the extnded radius of the tooth (from the base cylinder) [only if using default tooth profile]
    toothBottom - Z size of base of tooth [only if using default tooth profile]
    toothTop - Z size of end of tooth [only if using default tooth profile]
    external - if True makes an external thread (external to base cyldiner). Otherwise, an internal thread.
    starts - number of thread starts
    neckInDegrees - degrees used to feather bottom toothBite
    neckOutDegrees - degrees used to feather top toothBite
    segments - number of circle segments ($fs in OpenSCAD). More is smoother, but takes longer to render.
           
    The default thread profile will produce either a isosceles triangle profile (if tooothTop == 0), 
    or an isosceles trapezoid profile (if toothTop > 0); this is a triangle with a truncated (flat) top.

    Re pitch and lead:
        pitch (Z distance between threads) == lead / starts, and,
        lead (Z distance per turn) == pitch * starts
    
    '''
        
    if tooth == None:
        # if toothTop==0, this gives a triangular thread. If toothTop>0, gives a thread with a flat (truncated) top.
        tooth = [[0,         -toothBottom / 2],
                 [toothBite, -toothTop    / 2],
                 [toothBite, +toothTop    / 2],
                 [0,         +toothBottom / 2]]
       
    thread = []
    
    for twist in np.linspace(0,360,num=starts,endpoint=False).tolist():
        
        start = screw_thread.thread(outline_pts=tooth, 
                                inner_rad=diameter/2,
                                external=external,
                                pitch=lead, 
                                length=length, 
                                segments_per_rot=segments,
                                neck_in_degrees=neckInDegrees, neck_out_degrees=neckOutDegrees)
        
        start = rotate([0,0,twist])(start)
        
        thread += start
    
    return thread`

etjones, 2018-07-25:
Sweet!  There's more detail in the thread standard than I can quite grok, but I'd love it if the screws `screw_thread.thread()` generated were in line with the standard.  Even more if we had some presets so we could just say, e.g. `socket = screw_thread.quarter_twenty(external=False, height=25)`

What would you like to do with this?  It wouldn't take much code to make `screw_thread.thread()` take a `starts` argument; I could pop that in quickly.  I also think that the flat topped thread profile is a good idea; I'd like to make that the default.  I've got to do a little more looking at the standard to figure exactly where the flat segment of the default thread profile should go, but I think that's worth adding.

nerdfever, 2018-07-25:
The thread standard is simpler than it looks.

I've already done what I wanted to with this - I thought you might like the code.

Nice as it sounds in theory to implement standard threads (and it's not difficult in SolidPython; with the right parameters the existing code will do it), in practice FDM-type printers can't print them accurately enough to be useful (at least not on screws smaller than my fingers). Maybe someday that'll change.