SolidCode/SolidPython #101: problem with object tree; a bug or a feature?

nickc92, 2019-01-06:
So I've encountered an issue which I don't know is a bug or a feature; the following code fails:
```
a = cube()
b = a + translate([1,0,0]) (a)
c = b + translate([0,1,0]) (b)
print(scad_render(c))
```

The reason seems to be due to what I think is an optimization: on line 3, we are creating a union of `b` with `translate([0,1,0]) (b)`.  But `b` itself is a `union()` object, so the `__add__` magic method of the `union` class is being called.  I'm guessing this is an optimization to prevent things like `union() { union() { union() {...` appearing in the rendered OpenSCAD code, but it has the unfortunate effect of making code like mine above fail.  By commenting out the `__add__()` method of the `union` class, and letting things fall back to `OpenSCADObject.__add__()`, things seem to work.  I don't know if this problem exists with other classes besides `union`, but it may.

Although the above code is contrived to make a simple example of the problem, it comes from real use cases; I've encountered it twice now.  One can get around the problem by working in the functional, OpenSCAD style:
```
def a(): return cube()
def b(): return a() + translate([1,0,0]) (a())
c = b() + translate([0,1,0]) (b())
print(scad_render(c))
```
But I wasn't sure if it was intended for the user to be forced into this functional style.

nickc92, 2019-01-15:
No thoughts?

nickc92, 2019-01-15:
You're right- it would be nice to have the cake and eat it too; it would be great if you could keep the optimization and avoid the problems it is causing.  Let me think about it too; though truth be told, it's one of those problems that makes my head hurt!

nickc92, 2019-01-15:
Perhaps one thing that might work is if the __add__() method of union could scan the set of all child nodes, and in the event that one of the children == self, then it creates a new union object; otherwise, it acts as it is coded right now.  It seems like this would prevent an OpenSCADObject from being a child of itself, which I think is what is causing the trouble with the infinite regress.

nickc92, 2019-01-15:
So I think we can have our cake and eat it too.  I replaced the `union.__add__()` method with this:
```
def __add__(self, x):
        new_union = union()
        for child in self.children:
            new_union.add(child)
        new_union.add(x)

        return new_union
```
And I think it accomplishes what we want.  I think a fundamental problem with the way `union.__add__()` was before is that it modified the object itself.  So if `b` is a `union()` object, and we say `c = b + b`, it modifies the `b` object; but that is not really what is intended I think.
Give it a think, and let me know what you think! I created a new pull request for this issue.

nickc92, 2019-02-11:
Did anyone try my suggested fix, with the simple replacement of `union.__add__()` like above?  I've been using SolidPython with this fix for a while, and I haven't run into problems.

etjones, 2019-02-11:
You're right. Just merged. Thanks!

closed 2019-02-11 by etjones