SolidCode/SolidPython #104: Can't add special variables like "$fn" from python code

tatic0, 2019-02-19:
Hello,

I can' t find a way to pass the value of special variables [1][2] directly from the python script, I have to re-edit the scad file manually.

I'd like to have to possibility to pass "$fn=360", for example.

[1] http://www.openscad.org/cheatsheet/
[2] https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#.24fa.2C_.24fs_and_.24fn

etjones, 2019-02-19:
This is definitely a hole in the documentation!  The most common case is, as you say, `$fn`. For that, you can specify a `segments` argument for any function (`sphere()`, `cylinder()`, etc.)  that would take `$fn`.  You can also check out the pattern in the [solidpython_template example](https://github.com/SolidCode/SolidPython/blob/master/solid/examples/solidpython_template.py) where you speficy `SEGMENTS` at the top of your SolidPython file, and add that as a header to the rendered SCAD code.  This lets you control the segments for the entire model in one place.  I've found this the easiest way to work for quick renders (`SEGMENTS = 12`) which I can increase for really smooth curves once I'm ready to render out (`SEGMENTS = 120`). 

I'll open an issue specifically to document OpenSCAD special variables and how to deal with them.

closed 2019-02-19 by etjones

nickc92, 2019-02-19:
So the $fn variable can be set on individual objects using the “segments” keyword, e.g.:
sphere(1.0, segments=50)
But I don’t believe there is a way to set $fn globally (Evan, maybe you can chime in if I’m wrong). 
It may be worth mentioning that the “viewscad” module, which lets you render solidpython stuff without the openscad GUI, allows you to set those variables: 
renderer.render(obj, dollar_sign_vars={‘fn’: 50})
-Nick

etjones, 2019-02-19:
You're correct that there's not a programmatic way to set $fn globally. Maybe there should be? 

Here's the pattern I use (and it's in the [solidpython_template example](https://github.com/SolidCode/SolidPython/blob/master/solid/examples/solidpython_template.py) ) to specify $fn globally:

```python
SEGMENTS = 12
a = assembly() # some SolidPython object
scad_render_to_file(a, file_header='$fn = %s;' % SEGMENTS)

```