SolidCode/SolidPython #109: Namespaces needed for imported scad code.

etjones, 2019-05-08:
We can currently import & use OpenSCAD code with `use(<SCAD_PATH>)`. This creates stub OpenSCAD objects for all detected modules and functions... and inserts them in the importing module's namespace. If you want to include two SCAD files with the same named function (`gear()`, say, which I ran into today), you're out of luck. 

A more robust way of doing this might be to create a statement like `<namespace> = import_scad(<SCAD_PATH>)`  that would put everything in a Python namespace. But... how would the compiled SCAD code work? I'm not sure OpenSCAD has a solution for this.  Looking at [this issue](https://github.com/openscad/openscad/issues/522), it seems like OpenSCAD is still as broken as SolidPython in this respect. Hm...

rockstorm101, 2019-08-17:
+1. I would love to see namespaces implemented too.

etjones, 2019-08-18:
I built most of this in a branch a couple months ago. Just need to polish it off and merge to master. I think OpenSCAD doesn't have any namespacing, but I think the more Pythonic behavior of namespaced imports would make better code. So... it's in the works, but not finished yet.

rockstorm101, 2019-08-19:
Hi, thanks a lot for responding. I just found this library I think it is great. Good to see you are working on this. The lack of namespaces in OpenSCAD is something that'd be great to overcome with SolidPython.

Happy to do testing or whatever you need. Just let me know.

etjones, 2019-08-31:
So, I just pushed a branch, [feature/scad_import_namespaces_gh109](https://github.com/SolidCode/SolidPython/tree/feature/scad_import_namespaces_gh109), with the following new pattern: 

```python
import solid
scad_mod = solid.import_scad('/path/to/scad_file.scad')
openscad_obj = scad_mod.your_mod(1,2,3)
```

It also recursively loads _all_ the scad files in a directory if that's specified, so:

```python
import solid
mcad = solid.import_scad('/path/to/MCAD')
motor_mount = mcad.motors.stepper_motor_mount(17)
```

Does that seem like about what you'd expect? I'm wondering a little bit if it's worth pursuing some deeper Python import mechanisms (say, `from import_scad('/path/to/MCAD') import motors`?) but I'm not sure if that would be more useful or just more confusing. Any suggestions?

etjones, 2019-09-03:
Resolved in v0.4.0, ea9411ff6f5d95fcb1fb00476b07fa9996d5ba33

closed 2019-09-03 by etjones