SolidCode/SolidPython #112: Try open this file, preview or render failed, what's wrong??

jumbojing, 2019-06-14:
def thread(P=1, #螺距,
           length=2, #螺丝的总长,
           Rmaj=5, #外径,
           sn = 36):

    angle = 360/sn

    Pt = []

    for i in range(sn):
        if i > round(sn / 2):
            i = i - round(sn / 2)
        r = Rmaj - i / sn * P
        p = [r * cos( i* angle), r * sin(i * angle)]

        Pt.append(p)

    c = P2(Pt)
    c = Le(height = length, convexity = 10, twist = -360.0*length/P, 
           center = True, slices = 500)(c)

c = Pr()(Ry(90)(thread()))
scad_render_to_file(c,"testcc.scad")

Try to open the testcc.scad, but preview or render failed, what's wrong??

this is testcc.scad:

projection() {
	rotate(a = [0, 90, 0]) {
		linear_extrude(center = true, convexity = 10, height = 2, slices = 500, twist = -720.0000000000) {
			polygon(paths = [[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]], points = [[5.0000000000, 0.0000000000], [-4.1720501029, -2.7049938569], [2.0177390834, 4.5140070730], [0.7584029619, -4.8578221518], [-3.2605860792, 3.6427754512], [4.6908070829, -1.2754333166], [-4.6033294053, -1.4732513353], [3.0434506148, 3.7189746642], [-0.5274057206, -4.7485791243], [-2.1283496766, 4.2464841521], [4.0720613414, -2.3911710830], [-4.6898477069, -0.2076947943], [3.7995111958, 2.7095188597], [-1.7038236718, -4.3146581578], [-0.9121403690, 4.5199939852], [3.2048995297, -3.2765169691], [-4.4445335361, 0.9996039548], [4.2470317943, 1.5695517013], [-2.6930703108, -3.6051868608], [-4.1720501029, -2.7049938569], [2.0177390834, 4.5140070730], [0.7584029619, -4.8578221518], [-3.2605860792, 3.6427754512], [4.6908070829, -1.2754333166], [-4.6033294053, -1.4732513353], [3.0434506148, 3.7189746642], [-0.5274057206, -4.7485791243], [-2.1283496766, 4.2464841521], [4.0720613414, -2.3911710830], [-4.6898477069, -0.2076947943], [3.7995111958, 2.7095188597], [-1.7038236718, -4.3146581578], [-0.9121403690, 4.5199939852], [3.2048995297, -3.2765169691], [-4.4445335361, 0.9996039548], [4.2470317943, 1.5695517013]]);
		}
	}
}

nickc92, 2019-06-14:
I think the main issue here is that you are trying to use degrees in the python `sin()` and `cos()` functions.  OpenSCAD uses degrees everywhere, but the python math module trig functions use radians.  Try changing the line `angle = 360/sn` to `angle = 2 * pi / sn`.

jumbojing, 2019-06-14:
@nickc92 Great!Successful!thank you very much!!
And another question, how to measure the point-to-point distance in openscad or solidpython, is there a good way?

etjones, 2019-06-14:
SolidPython includes  a vector math library, `euclid3`.  If you work with `Vector3`s, `Point3`s, etc, you should be able to measure vector lengths with `.magnitude()`

Ex:
```python

from euclid3 import Vector3
a = Vector3(1,0, 0)
b = Vector3(2,1,2)
c = a - b
print(d.magnitude())
>> 2.449489742783178

```
Good luck!

jumbojing, 2019-06-16:
thanks,but i wanna snap (like autocad)special points, and then measure...

etjones, 2019-06-16:
Sorry for the bad news, but that’s one of the hard limits of OpenSCAD; there’s no way to get input from the rendered geometry back to the program that generated it. You may have better luck with a more full-featured CAD program.

jumbojing, 2019-06-22:
The python code:
```
from solid import *
from solid.utils import *
import viewscad
r = viewscad.Renderer()
def thread(P=1, #螺距,
           length=10, #螺丝的总长,
           Rmaj=10, #外径,
           sn = 36):

    angle = 2 * pi/sn

    Po = []

    for i in range(0,sn):
        j = i
        if i > round(sn / 2):
            j = sn - i
        r = Rmaj - j / sn * P
        p = [r * cos( i* angle), r * sin(i * angle)]

        Po.append(p)

    Fa = polygon(Po)
    p3 = linear_extrude(height = length, convexity = 10, twist = -360.0*length/P, center = True)(Fa)

    print(scad_render(c))
    # r.render(c)
#     print(Pt)
    return(p3)
c = thread()
# c = Le(.1)(Pr()(Ry(90)(c)))
r.render(c)
```
-------
The scad code:
```
linear_extrude(center = true, convexity = 10, height = 10, twist = -3600.0000000000) {
	polygon(paths = [[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]], points = [[10.0000000000, 0.0000000000], [9.8207217592, 1.7316582162], [9.3447210623, 3.4012003142], [8.5880852542, 4.9583333333], [7.5753283820, 6.3564552513], [6.3386000400, 7.5540493696], [4.9166666667, 8.5159164705], [3.3536975165, 9.2142081983], [1.6978932927, 9.6292313628], [0.0000000000, 9.7500000000], [-1.6882461718, 9.5745198210], [-3.3156952784, 9.1097979071], [-4.8333333333, 8.3715789032], [-6.1957583489, 7.3838172712], [-7.3625382589, 6.1779031375], [-8.2994101196, 4.7916666667], [-8.9792850431, 3.2681924807], [-9.3830294245, 1.6544812483], [-9.5000000000, 0.0000000000], [-9.3830294245, -1.6544812483], [-8.9792850431, -3.2681924807], [-8.2994101196, -4.7916666667], [-7.3625382589, -6.1779031375], [-6.1957583489, -7.3838172712], [-4.8333333333, -8.3715789032], [-3.3156952784, -9.1097979071], [-1.6882461718, -9.5745198210], [-0.0000000000, -9.7500000000], [1.6978932927, -9.6292313628], [3.3536975165, -9.2142081983], [4.9166666667, -8.5159164705], [6.3386000400, -7.5540493696], [7.5753283820, -6.3564552513], [8.5880852542, -4.9583333333], [9.3447210623, -3.4012003142], [9.8207217592, -1.7316582162]]);
}
```
I use ViewSCAD to render a SolidPython object in Jupyter: 

[image]

The height is about 5,why???

jumbojing, 2019-06-22:
I load the scad code in the OPENSCAD IDE:

[image]

The height is 10

nickc92, 2019-06-22:
> 屏幕快照 2019-06-22 上午11 41 15

Hi, when I tried your code, it looks like I got something which is 10
units high.  One thing to mention is that the grid squares are not
always 1.0 in size.  It depends on the size of the rendered object. 
I've pasted my render below, and at the bottom of the image, there is a
square with "=2.0" written next to it.  This indicates that the square
is 2.0 in size.
-Nick

jumbojing, 2019-06-22:
@

> Hi, when I tried your code, it looks like I got something which is 10 units high.  One thing to mention is that the grid squares are not always 1.0 in size.  It depends on the size of the rendered [...]

Well, I have realized this, but thank you for your prompt answer.

closed 2019-06-24 by etjones