SolidCode/SolidPython #113: import_stl and import_dxf are deprecated

rockstorm101, 2019-08-17:
The use of `import_stl` is now deprecated in favor of just `import` [[1]].

[1]: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Importing_Geometry#import_stl

etjones, 2019-08-18:
Do you think we should remove `import_stl()` and `import_dxf()`? I think what happens when either are used in SP code is that the generated OpenSCAD code will print a deprecation warning. That seems like the right behavior until OpenSCAD actually removes the functions. 

The other option would be to include a deprecation warning _in the Python_ so you didn't have to open the generated code in OpenSCAD to get a warning. I'd take a pull request for that behavior if you submitted it.

rockstorm101, 2019-08-19:
Please excuse my brevity on this one. My complaint when opening the issue was that no `import` function was provided. But I just realised it is indeed provided, it is just renamed to `import_` (I'm guessing to avoid clashing with python's own `import`).

If you intend to remove the `import_stl` and `import_dxf` once they are removed from OpenSCAD then I would suggest adding a deprecation warning on SolidPython too. Leave this issue open until then or close if you wish, I'm happy with both options.

Otherwise if you intend to keep that backwards compatibility, I would close this issue as it is no longer relevant.

etjones, 2019-08-19:
Cheers. I'll close now and revisit when OpenSCAD actually makes some changes.

closed 2019-08-19 by etjones