SolidCode/SolidPython #128: ViewSCAD SVG inline support

Gahia123, 2019-11-05:
I've installed solidpython (0.4.4) and viewscad (0.2.0) via pip, openscad (2019.05-3) and firefox (70.0-1.1) in linux:
`pip install --user install solidpython viewscad`
I run a simple code:
`python main.py`
And throw me the attached error related with SVG support, I reported it here becouse ViewSCAD git hasn't aviability of report bugs.

[ERROR.txt](https://github.com/SolidCode/SolidPython/files/3806972/ERROR.txt)
[mainpy.txt](https://github.com/SolidCode/SolidPython/files/3806978/mainpy.txt)

etjones, 2019-11-05:
Did you have a Jupyter notebook running? Viewscad is meant to be used inside a Jupyter notebook, and it looks like the results you got were from running the `mainpy.txt` file directly from the command line. 

If you go back to the same environment where you got the errors before and try running `jupyter notebook` on the command line, a Firefox window should launch showing a list of files. If you look in that page at the upper right and choose the "New" button, it will give you the option to start a new "Python 3" notebook. Do that, and another Firefox tab will open where you can enter the contents of `mainpy.txt` and then press the "Run" button. I think that should just work for you; it did for me. 

I'm not sure if you need Jupyter notebooks or not. If you just want to use SolidPython with the OpenSCAD app, things will be even simpler. Let me know if you have any problems and I can talk you through a little bit more.

Gahia123, 2019-11-06:
I've made the abovesaid and the output is this:
`VBox(children=(HTML(value=''), Renderer(background='#cccc88', background_opacity=0.0, camera=PerspectiveCamera…`
 For now I'll use the OpenSCAD app, as a recomendation I think that make ViewScad module `jupiter` independent will be simpler for people that want to preserve their workflow: terminal emulator and Vim or others. SolidPython is a nice software, it adds functions in Python that OpenSCAD lackes; thank you.

etjones, 2019-11-06:
Good luck. Enjoy!

closed 2019-11-06 by etjones