SolidCode/SolidPython #133: Useful constants, render(), ring(), squorus(), label(), thread(), knurl()

nerdfever, 2019-12-30:
Maybe useful to somebody. Just run it and look at the resulting .scad to see what it can do.

Redistribution is welcome (free).

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""
    Nerdfever.com personal Python library for SolidPython.

"""

__author__    = 'NerdFever.com'
__copyright__ = 'Copyright 2018-2019 NerdFever.com'
__version__   = ''
__email__     = 'dave@nerdfever.com'
__status__    = 'Development'
__license__   = """Copyright 2018-2019 NerdFever.com

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

       http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
"""

'''

Modifiers:

.set_modifier("*") # disable aka disable()
.set_modifier("!") # show only aka root()
.set_modifier("#") # highlight aka debug()
.set_modifier("%") # transparent aka background())

'''

import os, math
import numpy as np
from solid import *
from solid.utils import *
from solid import screw_thread
#from nerdlib import *

# controls number of segments in rendered circles (==$fn in OpenSCAD)
SEGMENTS = 128

''' CONSTANTS '''

ZortraxM200BuildVolume = [200, 200, 180] # x, y, z, max

# these work on Z-ABS at layer size 0.14 to 0.19 mm (at least - haven't tested enough otherwise)
# all are per side, use double for diameters

# NOTE - These work at 0.19 mm/layer, normal quality.
#        At 0.09mm/layer + high quality, they are all a little loose.

TightFit                    = 0.1875 # considerable resistance - stays put
SnugFit                     = 0.25   # virtually zero contact
LooseFit                    = 0.50
VeryLooseFit                = 0.75

# Major (outer) diameters - ref https://en.wikipedia.org/wiki/Unified_Thread_Standard
Screw_0_80_Major_Diameter   = 1.5240
[... 198 lines cut ...]
        demo( squorus(3, [10, 20, 5], center=False) )

        demo( squorus(3, 10, center=True) )

        demo( squorus(3, [10, 20, 5], center=True) )

        demo ( label("""Hello\nworld!"""))

        demo( thread(diameter=5, length=10, lead=3,
                     toothBite=2,
                     toothBottom=1,
                     toothTop=1,
                     starts=2,
                     neckInDegrees=180,
                     neckOutDegrees=60,
                     external=True) )

        demo( knurl(d=5, h=3) )

        demo( knurl(d=5, h=3, center=True) )

        render(thing)

    main()
```

etjones, 2019-12-31:
This is great! I'd like to incorporate a couple of these things into `solid.utils`. Doable?  
I'm, uh, blissfully ignorant about the fine points of licensing, so please excuse a couple questions.

If I were to put, say, `label()` into `solid.utils`, is there particular attribution you'd want outside the docstring? The rest of SolidPython is all LGPL 2.1.  When people make contributions, I usually put a name and link in the docstring and call it good, but you've already got all your metadata set up-- what do you think is the best way to combine those two?

nerdfever, 2019-12-31:
Totally doable - that's why I posted it here (I'm not really into the whole git hacking thing).

Just cut-and-paste whatever you like - I don't mind and hereby give permission.

Make changes if you like, too. Yesterday I made a few minor improvements (mostly in label). It's frustrating that we don't know exactly what all the text() parameters really do except by trial-and-error.

Latest version (today) is below:

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""
    Nerdfever.com personal Python library for SolidPython.

"""

__author__    = 'NerdFever.com'
__copyright__ = 'Copyright 2018-2019 NerdFever.com'
__version__   = ''
__email__     = 'dave@nerdfever.com'
__status__    = 'Development'
__license__   = """Copyright 2018-2019 NerdFever.com

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

       http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
"""

'''

Modifiers:

.set_modifier("*") # disable aka disable()
.set_modifier("!") # show only aka root()
.set_modifier("#") # highlight aka debug()
.set_modifier("%") # transparent aka background())

'''

import os, math
import numpy as np
from solid import *
from solid.utils import *
from solid import screw_thread
#from nerdlib import *

# controls number of segments in rendered circles (==$fn in OpenSCAD)
SEGMENTS = 128

''' CONSTANTS '''

ZortraxM200BuildVolume = [200, 200, 180] # x, y, z, max

# these work on Z-ABS at layer size 0.14 to 0.19 mm (at least - haven't tested enough otherwise)
# all are per side, use double for diameters

# NOTE - These work at 0.19 mm/layer, normal quality.
#        At 0.09mm/layer + high quality, they are all a little loose.

TightFit                    = 0.1875 # considerable resistance - stays put
SnugFit                     = 0.25   # virtually zero contact
LooseFit                    = 0.50
[... 202 lines cut ...]
        demo( squorus(3, [10, 20, 5], center=False) )

        demo( squorus(3, 10, center=True) )

        demo( squorus(3, [10, 20, 5], center=True) )

        demo( label("""Hello\nworld!""") )

        demo( thread(diameter=5, length=10, lead=3,
                     toothBite=2,
                     toothBottom=1,
                     toothTop=1,
                     starts=2,
                     neckInDegrees=180,
                     neckOutDegrees=60,
                     external=True) )

        demo( knurl(d=5, h=3) )

        demo( knurl(d=5, h=3, center=True) )

        render(thing)

    main()
```

etjones, 2019-12-31:
Right on. Free use is best use!