SolidCode/SolidPython #136: Library not found: solid.splines

wieman01, 2020-03-18:
I am trying to generate a model using Bezier lines as described in one of your tutorials:

from solid.splines import catmull_rom_polygon, bezier_polygon

I installed the latest version of SolidPython from Github, but unfortunately VS Code won't find the "splines" package. Whenever I run my Python program I get this error message:

`ModuleNotFoundError: No module named 'solid.splines'`

This could be due to a user error. Any idea?

Any big thanks for this great tool!

etjones, 2020-03-18:
I think the problem is that we haven't done a PyPI release including the splines code yet. If you did `pip install solidpython`, you wouldn't get that module. That's worth correcting; I'll do a release later today. In the interim, you could try (in a new virtualenv) installing straight from Github:  `pip install git+https://github.com/SolidCode/SolidPython.git`

That works OK with the spline example.

wieman01, 2020-03-18:
Hello, 

Ah... my mistake. I did install the latest version from the GIT repository, but missed to remove the old package I had installed via PIP.

I removed SolidPython through "pip uninstall solidpython" and reinstalled the GIT version, that did the trick. VS Code not finds the package.

Thank you very much for your quick response, Evan!

etjones, 2020-03-18:
Glad it worked out. Happy CADding!

closed 2020-03-18 by etjones

etjones, 2020-03-18:
I just published v0.4.7 to PyPI, which I've confirmed contains the spline code. So `pip install solidpython` should now work with splines

wieman01, 2020-03-19:
Evan, 

Thank you for publishing this. It's great work and your curve library is fantastic, exactly what I have been looking for. It's a life-saver.

etjones, 2020-03-19:
That's great to hear. Thanks!