SolidCode/SolidPython #139: How to use Bezier_points and Bezier_polygon ?

Frank-bool, 2020-04-15:
In my job, I need to use  bezier curve. But I reported an error after compiling. So how do I use the Bessel function? How do I give the coordinates?

`b_points = solid.splines.bezier_points([[0,0], [2,6],[5,10],[15,5]], 10)`

Error：
`
  File  line 18, in <module>
    b_points = sd.splines.bezier_points([[0,0], [2,6],[5,10],[15,5]], 10)
  File "D:\Anaconda\lib\site-packages\solid\splines.py", line 162, in bezier_points
    points.append(_point_along_bez4(*controls, u))
  File "D:\Anaconda\lib\site-packages\solid\splines.py", line 166, in _point_along_bez4
    x = _bez03(u)*p0.x + _bez13(u)*p1.x + _bez23(u)*p2.x + _bez33(u)*p3.x
AttributeError: 'list' object has no attribute 'x'`

I think it's a question of giving coordinates. So how do I give the coordinates correctly?
Thank you for your help.

etjones, 2020-04-15:
That's a great question, Frank. It looks like `bezier_points()` requires a list of Point2 objects, and that's what's causing the errror. That shouldn't be the case, and I'm going to fix `bezier_points()` so it accepts a list of lists like you have. 

For the moment, though, you can call the function `solid.utils.euclidify()` to change that list of x,y pairs to Point2 objects.

etjones, 2020-04-15:
Here's a demo program that works right now. I'll add a comment when I release the fix later today: 

```python
#! /usr/bin/env python

from solid import polygon, scad_render_to_file
from solid.splines import bezier_points
from solid.utils import euclidify

SEGMENTS = 48

def bezier_test():
    points_list = euclidify([[0,0], [2,6],[5,10],[15,5]])
    b_points = bezier_points(points_list, 10)
    poly = polygon(b_points)

    # This will throw an exception until I put a fix in. Should work
    # in the next release of SolidPython
    # b_points_list = bezier_points([[0,0], [2,6],[5,10],[15,5]], 10)

    return poly

if __name__ == '__main__':
    poly = bezier_test()
    scad_render_to_file(poly, file_header='$fn = %s;' % SEGMENTS, include_orig_code=True)
```

closed 2020-04-15 by etjones

etjones, 2020-04-15:
Fixed in v0.4.8, now available from PyPi with `pip install solidpython`

commit 9bd8df5c (2020-04-15): Updated to v0.4.8
Resolves #139; Spline functions now accept sequences of sequences of floats (e.g. [(1,2), (3.2, 4)] instead of sequences of Point23s