SolidCode/SolidPython #140: pyinstall incompatibility

dennisjs, 2020-04-23:
Bug description:

This isn't exactly a bug, but solidpython is currently incompatible with pyinstaller builds. When building with pyinstaller, issues arise with 'scad_render_to_file'.

Note that I found a workaround: copy the 'pyproject.toml' file to the resultant exectuable's folder. Also, 'include_orig_code = False' in scad_render_to_file, as this portion of the library also fails during the compile.

Version:

0.4.8

Python Version: 3.8.2

Explain how the maintainer can reproduce the bug:

compile code with solid includes using pyinstaller
include a scad_render_to_file call

Expected behavior:

Successful code execution, including SCAD rendering

Actual behavior:

Code errors out with scad_render_to_file

Errors"

this line triggered the error:

file_out = sld.scad_render_to_file(cyl, out_dir=out_dir, file_header=f'$fn = {SEGMENTS};')

Traceback (most recent call last):
File "site-packages\solid\solidpython.py", line 564, in get_version
File "site-packages\pkg_resources_init.py", line 481, in get_distribution
File "site-packages\pkg_resources_init_.py", line 357, in get_provider
File "site-packages\pkg_resources_init_.py", line 900, in require
File "site-packages\pkg_resources_init_.py", line 786, in resolve
pkg_resources.DistributionNotFound: The 'solidpython' distribution was not found and is required by the application

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "stlmod_gui.py", line 653, in
File "site-packages\solid\solidpython.py", line 503, in scad_render_to_file
File "site-packages\solid\solidpython.py", line 572, in _get_version
File "pathlib.py", line 1227, in read_text
File "pathlib.py", line 1213, in open
File "pathlib.py", line 1069, in _opener
FileNotFoundError: [Errno 2] No such file or directory: 'C:\Users\dschis\PycharmProjects\stlmod\dist\stlmod_gui\pyproject.toml'

<traceback object at 0x0D5BB668>

etjones, 2020-04-23:
Hmm. I wasn't aware of Pyinstaller before, but I know Python packaging has always been a headache; it's great that this project is smoothing some of that out. 

Fortunately, the functionality that's breaking here is pretty ancillary to what SP does. It's trying to write the current SolidPython version to the generated files. Without knowing exactly how Pyinstaller does its work, I'm not sure how I'd fix the version lookup code, but it's pretty simple to make it not fail catastrophically. 

For the moment, I think it should be sufficient to call `scad_render_to_file()` with `include_orig_code = False`, like so:

`file_out = sld.scad_render_to_file(cyl, out_dir=out_dir, file_header=f'$fn = {SEGMENTS};', include_orig_code=False)`

I *think* that should be sufficient to work with the current code, even without moving the `pyproject.toml` file. 

I've just written a patch to avoid the failure entirely (at the expense of writing the SP version as `<Unknown>` in the generated files), but that will take a little testing. I'll let you know when I've got that checked in.

rockstorm101, 2020-04-23:
> pkg_resources.DistributionNotFound: The 'solidpython' distribution was not found and is required by the application

Hi, I've never used pyinstaller but looking at the errors here. Why is `pkg_resources` unable to find solidpython information when it's been bundled with pyinstaller? Does it work with the other python packages bundled like this? It seems to me bundling programs with pyinstaller breaks the pkg_resources.

Does pyinstaller provide an alternative with similar functionality?

> I've just written a patch to avoid the failure entirely (at the expense of writing the SP version as <Unknown> in the generated files), but that will take a little testing. I'll let you know when I've got that checked in.

I would suggest investigating replacing `pkg_resources` with the new `importlib` like suggested [here][1].

[1]: https://stackoverflow.com/a/56331414

etjones, 2020-04-23:
I sympathize with this idea; if something is worth fixing, it's usually worth fixing right. But... I think this may be deeper into the rabbit hole than I want to climb just yet.

Here are things that make me less inclined to jump into this hole:
* `importlib` is Python3.8+ only;  that would require an upgrade for all users. 
* I'm not at all sure what `pyinstaller` does to make its installs (or whether *that* would play nicely with `importlib`),  
* The `pyinstaller` use case is pretty far from SP's core functionality. I don't mind tweaking a little to allow people to use `pyinstaller` but I'm not super excited about the research it would take to do this right. 

PRs distinctly welcome; otherwise I'll probably just push my little hack, which makes me feel a little dirty.

dennisjs, 2020-04-23:
I just checked:  making the include_orig_code=False change with the pyproject.toml gone doesn't resolve the issue.  You still need to include the pyproject.toml.  These mitigations aren't too bad - just wanted to make you aware.

etjones, 2020-04-23:
Oh, yep. Just looked at that bit. We currently ask for the module version even if `include_orig_code` is False, which throws the error.  Fixed that in my working fix. Thanks for the update.

unrealisticdreamer, 2020-05-10:
Great job on the software, maintenance, and support all. 

I really appreciate it.

If needed, another work around is below. No need to move any files. Works on Python 3.7 and up.

---------------------------------------------------------------------------------------------------

obj = your solid python code;

filepath=r'insert full path to .scad file'            
f = open(filepath,"w+")
print (scad_render(obj), file=f)
f.close()

commit 9ca2eda7 (2020-05-14): Should resolve #140. Don't calculate SP version unless asked to include source code in generated file, and don't choke if we can't find the SP version
closed 2020-05-14 by etjones