SolidCode/SolidPython #148: Extrude along path in y direction path with face in XZ plane

BAleiHi, 2020-07-13:
I'm trying to create a square face in the XZ plan and extrude it along the y-axis and don't get the expected result.  Here's the code:
`from euclid3 import Point3

from solid import scad_render_to_file
from solid.utils import extrude_along_path

a = extrude_along_path(shape_pts=[Point3(0,0,0),Point3(0,0,2),Point3(2,0,2),Point3(2,0,0)], 
                       path_pts=[Point3(0,0,2),Point3(0,5,2),Point3(0,10,2)], scale_factors=None)
file_out = scad_render_to_file(a, 'test.scad', include_orig_code=True)`

Result in OpenSCAD is:
[image]

Thanks (please let me know if there is a better forum for this kind of inquiry)

BAleiHi, 2020-07-13:
Sorry - that was a test after I encountered the original problem.  The simplest version of this that did not work is:

from euclid3 import Point3
from solid import scad_render_to_file
from solid.utils import extrude_along_path

a = extrude_along_path(shape_pts=[Point3(0,0,0),Point3(0,0,2),Point3(2,0,2),Point3(2,0,0)], 
                       path_pts=[Point3(0,0,0),Point3(0,10,0)], scale_factors=None)
file_out = scad_render_to_file(a, 'test.scad', include_orig_code=True)

which produces:
[image]

Does extrude have limitations on how the "shape" and "path" are defined?

etjones, 2020-07-13:
The simple answer is that `shape_pts` need to lie in the XY-plane.  

If you change your line to:  
``` python 
a = extrude_along_path(shape_pts=[Point3(0,0,0),Point3(0,2,0),Point3(2,2,0),Point3(2,0,0)],
```
 ( transposing the Y & Z dimensions of your original) you end up with a rectangular prism as intended.

I need to check on this in the documentation. The XY-plane limitation is in the comments at the top of `solid.utils.extrude_along_path()`, but that doesn't help much if it doesn't make it into the docs, huh? Sorry it's been left out.

Anyway, for the moment, here's the comment from the `extrude_along_path()` source specifying limitations:

```python
    # Extrude the convex curve defined by shape_pts along path_pts.
    # -- For predictable results, shape_pts must be planar, convex, and lie
    # in the XY plane centered around the origin.
    #
    # -- len(scale_factors) should equal len(path_pts).  If not present, scale
    #       will be assumed to be 1.0 for each point in path_pts
    # -- Future additions might include corner styles (sharp, flattened, round)
    #       or a twist factor
```

As I think about it now, it should be possible to take the normal of `shape_pts` and make calculations based on that rather than assuming that `shape_pts` is in XY. I'll add this to the TODO queue

closed 2020-07-13 by etjones