SolidCode/SolidPython #151: extrude_along_path with scaling crashes: AttributeError: 'numpy.ndarray' object has no attribute 'x'

tpchuckles, 2020-07-18:
consider the following basic attempt at modelling a sheep's horn:

`from solid import * #SolidPython https://github.com/SolidCode/SolidPython
from solid.utils import *
import numpy as np

output="horn-SolidPython.scad"

thetas=np.linspace(0,np.pi,50)

path=list(zip(3*np.sin(thetas),3*np.cos(thetas),thetas))

profile=list(zip(np.sin(thetas),np.cos(thetas)))

scalepts=list(np.linspace(1.,.5,50))
#print(scalepts)

extruded=extrude_along_path(shape_pts=profile,path_pts=path,scale_factors=scalepts)

openscadCode="$fn=100;"
openscadCode=openscadCode+scad_render(extruded)
outFile=open(output,"w")
outFile.write(openscadCode)
outFile.close()`

run this (using python3 at least) and you'll encounter the following error:

`Traceback (most recent call last):
  File "horn.py", line 17, in <module>
    extruded=extrude_along_path(shape_pts=profile,path_pts=path,scale_factors=scalepts)
  File "/home/athena/.local/lib/python3.8/site-packages/solid/utils.py", line 1143, in extrude_along_path
    this_loop = [Point3(v.x, v.y, v.z) for v in this_loop]
  File "/home/athena/.local/lib/python3.8/site-packages/solid/utils.py", line 1143, in <listcomp>
    this_loop = [Point3(v.x, v.y, v.z) for v in this_loop]
AttributeError: 'numpy.ndarray' object has no attribute 'x'
`

it seems like the np objects we've created which we then attempt to scale dont have these x,y,zs:

`        if scale != 1.0:
            this_loop = [(scale * sh) for sh in shape_pts]
            # Convert this_loop back to points; scaling changes them to Vectors
            this_loop = [Point3(v.x, v.y, v.z) for v in this_loop]`

etjones, 2020-07-18:
I’m away from a computer right now, but I think this is an issue with expecting the profile path to be a list of tuples, lists, or `euclid3.Point3`s, but not expecting the `np.ndarray `s that your code uses. It’s reasonable and increasingly common to use numpy to create geometry, so I’ll take a look at adding this capacity. For the moment, i suspect (but haven’t verified) that the line:

``` Python
profile = list(zip(map(math.sin, thetas), map(math.cos, thetas))) 
```
wouldn’t show the same error. 

There’s one function, `solid.utils.euclidify() ` that would would make your original (quite graceful, btw) code work if updated. I’ll see what would be required to make that fix.

tpchuckles, 2020-07-18:
you're correct that the first code incorrectly presents tuples rather than euclid3.Point3 objects, but even when that is corrected (for i in range(N): profile[i]=Point3(profile[i][0],profile[i][1],profile[i][2])...etc, or via euclidify (thanks for that, that's awesome)), it still fails. 

they key to making it stop failing seemed to be in the building of scalepts list element-by-element. (I originally had posted a much different issue here, and since edited it, since this suggested the root of the issue was the numpy floats within the scalepts list). 

this code:

`1: path=euclidify(list(zip(3*np.sin(thetas),3*np.cos(thetas),thetas)))
2:	profile=euclidify(list(zip(np.sin(thetas),np.cos(thetas))))
3: 	scalepts=list(np.linspace(1,.1,N))

4: scalepts=[]
5: for i in range(N):
6:	scalepts.append((N-i)/N)
`

works with rows 4-6 present, fails with them commented out.

etjones, 2020-07-18:
Thanks, this is great info. I'll see what I can figure out

etjones, 2020-07-25:
Sorry for delay on this (I just had a baby! Little busy this week.) Looks like this is another numpy instance. Numpy has some different multiplication rules defined than standard python. Multiplying a standard python numeric by a `euclid3.Vector3 | euclid3.Point3` yields a `Vector3`.  Numpy's operator overrides supersede this, so multiplying a Numpy float64 by that same Vector3 yields a 3-tuple, which caused the problem you found. 

I'll push a fix for this behavior shortly, and then I'll see if I can add some more general behavior to `solid.utils.euclidify` that will play more nicely with Numpy.

etjones, 2020-07-25:
Pushed and released as V1.0.1

closed 2020-07-25 by etjones

etjones, 2020-07-25:
Also worth noting: although you could call `euclidify()` on all the numpy lines like `path=list(zip(3*np.sin(thetas),3*np.cos(thetas),thetas))`, you don't have to. Once the issue with `scalepts` was fixed, this just works.