SolidCode/SolidPython #153: How to read data of graphics.

Frank-bool, 2020-07-23:
Frist, thank you very much for the package, which is very helpful for my work.
I use this package to To create 3D-structures. If I want to create two structures and save them in one file, I need to determine whether the two structures intersect before I output the `.scad` file. Just like this:
```
A = difference()(
   cylinder(r=5, h=10),
   cube([4,4,4])
)
```
For example, these two structures, I built the cylinder and the cube. But I do not know whether these two structures intersect. I read the value of A. And the idea is, if A is empty, then there's no intersection, the opposite is the intersection. But I didn't succeed. I didn't find A way to read the value of A. So I would like to ask if there is A way to read out or determine if A is empty. 

Or is there a similar Python package that solves my problems and makes it easy to build the same structure as openSCAD? Your package is really great, I didn't find a package as easy to build structure as SolidPython.

If you have any solutions to my problems, please let me know. It will be of great help to me.

Thank you very much for your help and answer.

etjones, 2020-07-23:
I’m glad you’re enjoying SolidPython! The feature you’re looking for is something like introspection: once you’ve defined a shape, you want to find some information about it. This is a really useful feature... and unfortunately it’s not supported by OpenSCAD, and therefore not by SolidPython, which just generates OpenSCAD code. 

I’d love to hear some more about your particular problem. Sometimes if we reframe the issue we can work around the limitations of the software. 

Two other possible paths forward:

1) using OpenSCAD command line arguments, you could generate STL files directly from the generated OpenSCAD. If the STL is empty, you know the part intersection is empty. Note that this would require a certain amount of shell script glue and that OpenSCAD can take a long time to create STLs. Depending on your purposes, this might or might not be feasible for you. 

2) The open source 3D modeler Blender is extremely capable and has a Python API that could definitely do what you’re asking. I haven’t used it, but I have the impression that it’s a little more complicated to create geometry in Blender than SolidPython, since the API is so large and Blender is chiefly a polygon-based modeler rather than a solid-based modeler. 

Either way, let me know what direction you choose. Your issue isn’t uncommon and it’s helpful to others to be able to see what solutions worked for you. 

Cheers!

Frank-bool, 2020-07-23:
Thank you very much for your answers.

In the two ways that you mentioned:

1) My understanding is that you need to make a human judgment after generating the STL file. But my judgment of whether or not to cross the job is left to the computer to do, no human involvement. If the intersection, the output error, otherwise correct, output `.scad` file. 

2) You offer a good solution, and I'll look into that. I'll see if it's suitable for my current job and help me solve problems.

Thanks again for your help.

Cheers! ^_^

etjones, 2020-07-23:
Re: 1:

I assume you need a valid set of scad files?  What if you always generated the scad files (instant) and STL files (slow). I think a simple shell script could check the size of the STL and reject any below a certain size (the size of bare STL headers, maybe) and delete the associated .scad files for any runs that don’t make the cut. That would be a slow process, but would have the benefit of not requiring human intervention. 

However you decide, good luck!

Frank-bool, 2020-07-23:
Sounds like the first method is good.

It may be my personal problem, I don't quite understand your method. 
Maybe my question is a little silly, but can shell script operations be included in Python?

Because my idea is to let Python do all the work as much as possible.

Frank-bool, 2020-07-23:
According to the method you said, how can you achieve the function you said in Windows system? I don't know much about it. Could you please provide some information so that I can look up and learn?

etjones, 2020-07-23:
You can absolutely run shell scripts from Python. The simplest way is to call, e.g. os.system(‘/path/to/OpenSCAD -someflagToMakeStl stpath.stl yourfile.scad’)`

although it’s a surprisingly deep topic. A good intro is here: https://stackabuse.com/executing-shell-commands-with-python/

Things may be a little weird on Windows, because Windows, but definitely doable. `os.path.getsize()` would tell you the size of generated STLs. So... there’s a little hacking involved to get what you’re looking for, and I wouldn’t do it if I had less than 20 files to calculate, say, but there’s definitely a way forward

Frank-bool, 2020-07-23:
WoW! Your method is great, I will keep trying.

Thank you very much for your help! I apologize for the trouble I caused you.

closed 2020-08-04 by etjones