SolidCode/SolidPython #154: run_all_tests.sh

ottojas, 2020-07-26:
One failure on current cloned code: Running on debian  10 (buster)

test_solidpython.py
..................................E....................
Found OpenSCAD code that's not compatible with Python. 
Imported OpenSCAD code using `or` 
can be accessed with `or_` in SolidPython

Found OpenSCAD code that's not compatible with Python. 
Imported OpenSCAD code using `or` 
can be accessed with `or_` in SolidPython

.
======================================================================
ERROR: test_offset_zero_delta (__main__.TestSolidPython)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_solidpython.py", line 427, in test
    scad_obj = eval(call_str)
  File "<string>", line 1, in <module>
  File "/home/otto/.local/lib/python3.7/site-packages/solid/objects.py", line 423, in __init__
    raise ValueError("offset(): Must supply r or delta")
ValueError: offset(): Must supply r or delta

----------------------------------------------------------------------
Ran 56 tests in 1.185s

FAILED (errors=1)

closed 2020-07-27 by ottojas

etjones, 2020-07-28:
This sounded like a valid issue. Did you find some extra info about it that led you to close it?

ottojas, 2020-07-28:
It would have been an issue but, I was running tests against my pip
installed version.  I uninstalled the pip version and reinstalled it
and the newer version worked fine.  I have now stumbled at least twice
on git hub and am beginning to feel more and more exposed and less and
less competent. Oh well, so goes life (or at least programming!)

Regards
Otto

On Mon, 27 Jul 2020 17:17:06 -0700
Evan Jones <notifications@github.com> wrote:

> This sounded like a valid issue. Did you find some extra info about
> it that led you to close it?
>