SolidCode/SolidPython #156: Summary for curated list of code CAD

Irev-Dev, 2020-10-31:
Hey Evan 👋 ,

I finally got around to putting together a list of all of the code-CAD projects that I know of [here](https://kurthutten.com/blog/curated-code-cad), and also on [Github](https://github.com/Irev-Dev/curated-code-cad).

I've put a short summary of each one, but it's hard to do them all justice. for SolidPython I've only got:
Python-based library that wraps OpenScad, i.e. it outputs OpenScad code.

If you would like to give me a summary of the project, and perhaps some points of how it differs from other similar projects would be happy to include it. I think max three paragraphs would be about right. Open a PR on reply to this issue and I'll include it.

etjones, 2020-11-02:
Thanks for putting this together, Kurt. I think the phrase you've got is just fine; "Python in, OpenScad out" is the important message to get across, and your copy does that nicely. I appreciate you reaching out and giving me the opportunity to qualify as needed. Cheers!

closed 2020-11-02 by etjones

Irev-Dev, 2020-11-05:
Feel free to reach out if you change your mind, let me know if I can help you or SolidPython in anyway.

Irev-Dev, 2021-01-01:
Me again.

I also want to add some of the best real-world examples of Code-CAD being used out in the wild. Do you know of any good projects using SolidPython that I can feature?

So far what I've found is:
- [Acedemics using it for reactor design](https://github.com/ukaea/paramak)
- [Lego compatible blocks](https://fxbricks.com/)
- [specialised in-house equipment for drones](https://twitter.com/robsmallshire/status/1343477584458346497)
- [Pruse 3d printer](https://github.com/prusa3d/Original-Prusa-i3)