SolidCode/SolidPython #159: color() not working

saturn-hh, 2020-12-31:
First I would like to say thank you for the opportunity to work with solidpython and to do my first steps in CAD with solidpython. 

JuypterNotebook, viewscad and solidpython are up and running. Unfortunately I can't get my objects colored. I've tried 

`cube_color =  color(Red)(cube(10))
cube_blue = color([0,0,1, 0.5])(cube(5))

#r.render(cyls_2_debug)
r.render(cube_color)`

I get no error and the rendering is fine but the object has not the defined color. 

Any help is appreciated.

Alexander

Distribution: Linux Mint
solidpython 1.0.1
viewscad 0.2.0
python 3.8

etjones, 2020-12-31:
Hi! Good news- I recently pushed a couple builds that bring color handling into line with OpenSCAD's colors. It looks like your local version hasn't picked those up yet, so the first thing I'd recommend is:
`pip install --upgrade solidpython`

There *have* been some issues with OpenSCAD's own color handling in the past. If upgrading SolidPython to v1.0.4 doesn't fix your issue, you might try opening your OpenSCAD install and confirming that this OpensCAD code does in fact generate a red cube:  `color("red"){ cube();}`.  If you don't have an explicit OpenSCAD install, I think that ViewSCAD bundles it somewhere, and also that ViewSCAD might let you evaluate raw OpenSCAD (as opposed to SolidPython) code in Jupyter.

I'm betting that these should fix your issues, but please report back here if they don't and I'll dig in farther. Cheers!

saturn-hh, 2021-01-03:
Thank you for your comment.
I've tried your suggestion and got mixed results:
OpenScad as stand alone software rendered the example correct (cube in red)
viewscad doesn't render the openscad code at all. I get no result and no error message. So I think  the issue is related to viewscad. I'l ask there for a solution

Thank you again
Alexander

solidpython Version: 1.0.4
viewscad Version: 0.2.0

etjones, 2021-01-03:
Best of luck. The ViewScad author is really solid and should be a good help. 

One other thing you might try is using your local SolidPython install and confirming that it does actually correct OpenSCAD code, without ViewScad in the loop. If that works OK, then you can confirm that the issue is with ViewScad.

etjones, 2021-01-11:
Hey - Just saw your update about ViewScad. I just did a clean install of ViewScad and I also don't get color there. Here are my results: 

Locally, running OpenSCAD with SolidPython. Color shows up correctly: 
[image]

New virtualenv, with `pip install viewscad` and nothing else. Color not rendered: 
[image]

I don't know what the deal is with issues on the [Viewscad repo](https://github.com/nickc92/ViewSCAD).  The author @nickc92 has been active here before, so he may be interested in seeing this bug. Or, he may have just decided against supporting Viewscad. I think at this point it looks like a Viewscad issue more than a SolidPython one. Let me know if you run into any SolidPython-specific color issues, and I'll reopen this.

closed 2021-01-11 by etjones

kylekyle, 2023-10-15:
This, by the way, is because [STL doesn't support color](https://github.com/openscad/openscad/issues/4671) and ViewSCAD/JupyterSCAD render STLs in notebooks via pythreejs. OpenSCAD can export to other formats that support color, such as 3MF or DXF, but you would [need a way to render those in Jupyter](https://github.com/pschatzmann/jupyterlab-viewer-3d/issues/21). 

Or, if you want to stick to STL, you would have to [export each color separately](https://erik.nygren.org/2018-3dprint-multicolor-openscad.html), then used something like [K3D](https://k3d-jupyter.org/) to import each color into a single plot for rendering in Jupyter.