SolidCode/SolidPython #163: Fail due to (not really) missing positional arguments

rockstorm101, 2021-02-14:
The following SCAD file works just fine and generates a cube as expected:
```OpenSCAD
// my_module.scad

my_cube(arg3 = 5);

module my_cube(arg1, arg2, arg3)
{
    s = arg3
         ? arg3 * 2
         : arg2
             ? arg2 / 2
             : arg1;
    cube([s,s,s]);
}
```
However, importing such file into SolidPython and attempting the same result, i.e. the code below:
```Python
# test.py

import solid as sp

my_module = sp.import_scad('./my_module.scad')

def my_function():
    return my_module.my_cube(arg3 = 5)

if __name__ == '__main__':
    sp.scad_render_to_file(my_function(), include_orig_code = False)
```
Fails with the following error:
```shell
$ python test.py 
Traceback (most recent call last):
  File "/tmp/test/test.py", line 11, in <module>
    sp.scad_render_to_file(my_function(), include_orig_code = False)
  File "/tmp/test/test.py", line 8, in my_function
    return my_module.my_cube(arg3 = 5)
TypeError: __init__() missing 2 required positional arguments: 'arg1' and 'arg2'
```
Redefining `my_function` in `test.py` as per below to explicitly set 'arg1' and 'arg2' as 'None':
```python
def my_function():
    return my_module.my_cube(None, None, arg3 = 5)
```
Produces the SCAD code I was expecting:
```OpenSCAD
use </tmp/test/my_module.scad>

my_cube(arg3 = 5);
```
Happy to help debug/test further if required. 

System Info:
- SolidPython 1.0.5
- Python 3.9.1+
- Debian Sid

etjones, 2021-02-14:
This makes sense. It looks like I’ve taken Python’s args/kwargs differentiation and applied it to OpenSCAD, when in fact all arguments in OpenSCAD are optional. That should be a quick fix. Thanks for pointing this out.

etjones, 2021-02-14:
Fixed & tested in fa260864ce82bc6a4b8c72927ae307294b9dc566.  Thanks for including the test case; that made it really easy to confirm the fix

closed 2021-02-14 by etjones