SolidCode/SolidPython #164: Collision detection

kaolpr, 2021-02-20:
Do you have any handy method for detecting collision between bodies? I am thinking of a method for verification if elements of the assembly do not overlap.

etjones, 2021-02-20:
Unfortunately, OpenSCAD (and therefore SolidPython) has no capacity at all for introspection, meaning there’s no automated way to test for object intersection besides running OpenSCAD / CGAL and checking where your objects end up. 

This is a pretty fundamental limit on the program. When people ask about capabilities like this, I usually tell them they’ve outgrown OpenSCAD and point them to commercial CAD packages. One free place you might look is at Christopher Olaf’s ImplicitCAD (https://github.com/colah/ImplicitCAD). Its 3D kernel is based on very different algorithms than OpenSCAD/CGAL which can be a) much much faster than OpenSCAD, and b) more tightly integrated into the language, so there’s some possibility of introspectability. Give it a look!

kaolpr, 2021-02-22:
@etjones Thank you for explanation.
I'm also looking into CadQuery (https://github.com/CadQuery/cadquery), but have not yet established if what I need is possible there.

closed 2021-02-22 by kaolpr