SolidCode/SolidPython #180: native animations

jeff-dh, 2021-06-01:
```python
from solid import *

view_port_magic = """
                       $vpt = [4, 3, 15];
                       $vpr = [60, 0, 360 * $t];
                       $vpd = 100;
                   """

c = cube(scad_inline("$t*10"))

scad_render_to_file(c, file_header=view_port_magic)
```
yeah!

Works with expsolid should work with master too.

(you need scad_inline from #178)

etjones, 2021-06-01:
Oh shit, this could make animation so much better than the hack I have in place now!  Dude, you're improving all of this way faster than I can keep up. Nice work.

jeff-dh, 2021-06-01:
physics:

```python
import numpy as np

from solid import *

ball_radius = 100

def get_bouncing_ball_data(pos=np.array([0.0, 0.0, 0.0]), vel=np.array([5.0, 5.0, 200.0])):
    data = []
    gravity = np.array([0.0, 0.0, -8.0])
    for t in range(1000):
        vel = (vel + gravity) * 0.995
        pos += vel
        if pos[2] < ball_radius:
            vel[2] = vel[2] * -1
            pos[2] = ball_radius

        data.append(pos.tolist())

    return data

ball_data = f'bouncing_ball_data = {get_bouncing_ball_data()};'

ball_pos_over_time = scad_inline("bouncing_ball_data[$t * 1000]")

ball = translate(ball_pos_over_time)(sphere(ball_radius))
floor = background(cube([2000, 2000, 0.01]))

scad_render_to_file(ball + floor, file_header=ball_data)
```
No clue what this could be good for, but....... I guess it might be possible to plug a -- more complex :-p -- physics engine into it and render animated movies....

All of this is implemented and working with [exp_solid](https://github.com/jeff-dh/SolidPython/tree/exp_solid) and these snippets are in the [examples](https://github.com/jeff-dh/SolidPython/tree/exp_solid/solid/examples), check it out...

(but this should also work with master with little effort -> scad_inline from #178)

PS: anybody an idea where to get the vitamins for a 4d printer? :stuck_out_tongue: