SolidCode/SolidPython #187: extrude_along_path() is broken - normals inverted

ShurikMur, 2021-09-30:
After updating solidpython to the latest (1.1.1) the extrude_along_path function seems to get broken:
either first, last or side face of result solid gets a reversed normal and openSCAD crashes on rendering.
Changing the order of shape points reverses normals but there are still some faces facing wrong direction.

etjones, 2021-09-30:
Good catch on this. I was adding lots and lots of inverted facets to the `polyhedron()` calls, which I just now realized can be visualized with OpenSCAD's F12 'Thrown Together' view. Everything *should* look good now. You'd need to pull the latest master to see the fix, but it's there. I'm going to hold off on a release just now, because there's a much bigger release coming shortly and I'm not sure where this will end up in the mix.

Gigahawk, 2021-11-11:
As far as I can tell, this is still broken, as of 37417da, running `path_extrude_example.py` still breaks when trying to render in OpenSCAD.

etjones, 2021-11-11:
What kind of failure do you run into? Running the code at [37417da](https://github.com/SolidCode/SolidPython/commit/37417daac6397e7fdeab511136567536840f782a) it took me almost 14 minutes to render with CGAL, but everything looks like it's working correctly. What does your breakage look like?

Gigahawk, 2021-11-12:
Ok weird, I tried to recreate this in a clean environment and it seems to work fine, maybe pip didn't replace the regular 1.1.1 release when i tried to install from git.

My bad, feel free to close this again.

etjones, 2021-11-12:
All good- “no bug here” is the best outcome! Thanks for reporting.

closed 2021-11-12 by etjones