SolidCode/SolidPython #188: Cannot install in windows 10 using "pip install solidpython"

thephet, 2021-10-12:
You can see the trace here. It seems to require to install visual studio tools, but this is over 6gbs and I dont want to install it.

Is there any other alternative?

(CS) C:\Users\Juanma>pip install solidpython
Collecting solidpython
  Using cached solidpython-1.1.1-py3-none-any.whl (86 kB)
Collecting PrettyTable==0.7.2
  Using cached prettytable-0.7.2.zip (28 kB)
  Preparing metadata (setup.py) ... done
Collecting pypng<0.0.20,>=0.0.19
  Using cached pypng-0.0.19.tar.gz (293 kB)
  Preparing metadata (setup.py) ... done
Collecting regex<2020.0,>=2019.4
  Using cached regex-2019.12.20.tar.gz (679 kB)
  Preparing metadata (setup.py) ... done
Collecting euclid3<0.2.0,>=0.1.0
  Using cached euclid3-0.01.tar.gz (13 kB)
  Preparing metadata (setup.py) ... done
Using legacy 'setup.py install' for PrettyTable, since package 'wheel' is not installed.
Using legacy 'setup.py install' for euclid3, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pypng, since package 'wheel' is not installed.
Using legacy 'setup.py install' for regex, since package 'wheel' is not installed.
Installing collected packages: regex, pypng, PrettyTable, euclid3, solidpython
    Running setup.py install for regex ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\Juanma\CS\Scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Juanma\\AppData\\Local\\Temp\\pip-install-uu2y8x9w\\regex_d7af0b8bda4a4801bddf3a7d311ceadf\\setup.py'"'"'; __file__='"'"'C:\\Users\\Juanma\\AppData\\Local\\Temp\\pip-install-uu2y8x9w\\regex_d7af0b8bda4a4801bddf3a7d311ceadf\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Juanma\AppData\Local\Temp\pip-record-kk8gzn0b\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Juanma\CS\include\site\python3.9\regex'
         cwd: C:\Users\Juanma\AppData\Local\Temp\pip-install-uu2y8x9w\regex_d7af0b8bda4a4801bddf3a7d311ceadf\
    Complete output (13 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.9
    creating build\lib.win-amd64-3.9\regex
    copying regex_3\__init__.py -> build\lib.win-amd64-3.9\regex
    copying regex_3\regex.py -> build\lib.win-amd64-3.9\regex
    copying regex_3\_regex_core.py -> build\lib.win-amd64-3.9\regex
    copying regex_3\test_regex.py -> build\lib.win-amd64-3.9\regex
    running build_ext
    building 'regex._regex' extension
    error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\Juanma\CS\Scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Juanma\\AppData\\Local\\Temp\\pip-install-uu2y8x9w\\regex_d7af0b8bda4a4801bddf3a7d311ceadf\\setup.py'"'"'; __file__='"'"'C:\\Users\\Juanma\\AppData\\Local\\Temp\\pip-install-uu2y8x9w\\regex_d7af0b8bda4a4801bddf3a7d311ceadf\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Juanma\AppData\Local\Temp\pip-record-kk8gzn0b\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Juanma\CS\include\site\python3.9\regex' Check the logs for full command output.

jeff-dh, 2021-10-12:
The reason why regex depends on vs-blablub could be the following: I could imagine that the regex module is mostly a C/C++ implementation using a python wrapper for performance reasons. Furthermore it seems that you're pip command issued a "build" of regex and thus needs a compiler.

I'm not into the details of pip installations, but there's something about "wheel"(s) which -- I guess! -- are some kind of binary packages.
Your output says:

```
 Using legacy 'setup.py install' for regex, since package 'wheel' is not installed.
```
I would take a look at this wheel stuff and probably just try to install the "wheel" package, you might(!) be lucky to then be able to install a precompiled binary regex package without needing to install vs-blablub.

thephet, 2021-10-13:
I just did a "pip install regex" and it was fine. It installed 2021.10.8. Your project forces regex to be regex<2020.0,>=2019.4. I wonder if you could let it use the last version of regex, because then it'd install.

BTW thank you very much for your help.