SolidCode/SolidPython #189: subtractions of extrude_along_path() objects do not show correctly in 'Preview'.

asdfstuart, 2021-11-26:
Subtracting an extruded object from an OpenSCAD object or another extruded object seem to confuse the 'Preview' viewer.  The models render correctly and show correctly in the 'Surfaces' viewer.
I am using version 1.1.1 with the latest extrude_along_path.py containing the fixes for issue #18
I added the function extrude_example_subtractions() to the attached path_extrude_example.py file.  The problems show up most clearly if you rotate to look up the +y axis.

[path_extrude_example.zip](https://github.com/SolidCode/SolidPython/files/7610757/path_extrude_example.zip)
7.

etjones, 2021-11-27:
Thanks for including that function; it was really useful to see what was happening.

[image]

When I set the `convexity = 10` argument in the created polygons, the visualization issue is resolved.  Looking at the [`polyhedron()` docs](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron), it seems like `convexity=10` would be a more reasonable default value that wouldn't impact visualization speed too much. I'll leave this open until I've made a pass at that. 

[image]

asdfstuart, 2021-11-29:
Thanks for the quick fix, it worked for my model as well. Also, any convexity other than 1 worked for my model.
You are welcome for the function.  I really appreciate your extrude_along_path() function, it is useful and saves me a lot of time.  I am glad I could return the favor of  in a small way.