SolidCode/SolidPython #191: Plans for a new release?

yaqwsx, 2022-01-03:
Hello, I am sorry this is not really an issue rather than a discussion, but I don't have another way of contacting the active maintainers of this project.

Do you plan to have a new release? If so when? I use SolidPython in [yaqwsx/KiKit](https://github.com/yaqwsx/KiKit) and unfortunately, the latest release of SolidPython (v1.1.1) depends on the regex library that is troublesome on Windows (with the Python interpreter that KiCAD brings). Fortunately, you have [removed the dependency](https://github.com/SolidCode/SolidPython/commit/f7eb1536bd4ef98035c554f43ee001dba592d107), but since it is not released, I cannot enforce it from my Python package.

etjones, 2022-01-03:
Hi Jan - I think this is definitely an issue! I've delayed making a new release because there's a big re-write I've been working on. That's fallen behind a little, so it's probably best if I release anyway. Thanks for being in touch; I'll update this when there's a new release

etjones, 2022-01-03:
I just pushed a new release, v1.1.2, with no `regex` dependency. Should install fine with `pip install solidpython`.  Re-open this issue if you run into any problems. Cheers!

closed 2022-01-03 by etjones

yaqwsx, 2022-01-04:
Thanks a lot!