SolidCode/SolidPython #193: how can I reuse a nested transform?

maxlem, 2022-02-03:
Hello!

in openscad I could do
```
module my_tf()
{
    translate([post100_size_x-brace500_housing_depth-brace500_tenon_depth, brace500_width + brace500_housing_offset, to_brace503])
    rotate([90, 0, 0])
    children(0);
}

my_tf() cube();

my_tf() sphere();

```

I tried many things in python, notably

```
def my_tf():
    return translate([post100_size_x-brace500_housing_depth-brace500_tenon_depth, brace500_width + brace500_housing_offset, to_brace503])(
    rotate([90, 0, 0])(children(0)))

[...]

d = my_tf()(cube())
```

but it doesn't genereate the right code. Aside from computing a 4x4 matrix and sharing it, is there a syntax to achieve what I'm trying to achieve?

etjones, 2022-02-03:
I think what you're looking for is to use an explicit argument to `my_tf()`. OpenSCAD allows some implicit argument passing with `children()`, but that's harder to work out in Python. So I would do:

```python
    def my_tf(obj):
        post100_size_x = 1
        brace500_housing_depth = 2 
        brace500_tenon_depth = 3 
        brace500_width = 4 
        brace500_housing_offset = 5 
        to_brace503 = 6 

        x = post100_size_x - brace500_housing_depth - brace500_tenon_depth
        y = brace500_width + brace500_housing_offset
        z = to_brace503
        return translate([x, y, z])(
            rotate([90,0,0])(
                obj
            )
        )

```
You should then be able to call this with, e.g.  `my_tf( cube() )`

closed 2022-02-03 by etjones

maxlem, 2022-02-03:
I should have  found it myself, sorry, and thanks

etjones, 2022-02-03:
No worries! Happy modeling!