SolidCode/SolidPython #194: import_scad() does not look in 'Installation library' location

occivink, 2022-02-13:
Hi,

'Installation library' is the terminology used on the OpenScad wiki and described as:
"The Installation library location is the libraries directory under the directory where OpenSCAD is installed."
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries

So this `use <MCAD/2Dshapes.scad>` is valid in openscad, 
in SolidPython `import_scad("MCAD/2Dshapes.scad")` is not valid, it fails with the error
```
Locations searched were: [PosixPath('MCAD/2Dshapes.scad'), PosixPath('/home/user/.local/share/OpenSCAD/libraries/MCAD/2Dshapes.scad')]
```
indicating that the installation library path is not search at all.

This is occuring with solidpython 1.1.1 on arch linux

jeff-dh, 2022-02-13:
already mentioned in #176

(more or less a duplicate, even though the main issue in #176 is a different)

jeff-dh, 2022-02-13:
Btw: If you plan to use sophisticated OpenSCAD libraries, it might be worth to have a look at [expSolid](https://github.com/jeff-dh/SolidPython/tree/exp_solid), because there are several issues with OpenSCAD libraries in the master branch (i.e. #173, #176).

occivink, 2022-02-13:
Thanks for the hint. I was experimenting with dotScad but so far it seems to work as expected. I'll keep your warning in mind though

etjones, 2022-02-15:
Hi @occivink. Thanks for noting this. I wasn't aware of this extra library location. Is there a standard installation spot on Linux systems? On MacOS or Windows, that's pretty well defined, but it seems like different Linux distributions have different app layouts. If we have some solid guesses about where to try, I can make sure those paths are searched.

occivink, 2022-02-15:
On [Arch linux](https://archlinux.org/packages/community/x86_64/openscad/), [Ubuntu](https://packages.ubuntu.com/focal/all/openscad-mcad/filelist) and [Fedora](https://fedora.pkgs.org/35/fedora-aarch64/openscad-2021.01-4.fc35.aarch64.rpm.html) this path is `/usr/share/openscad/libraries`, so maybe this a rare case where distributions _can_ actually agree on something

etjones, 2022-02-15:
That's great. I'll add that to the search paths. Was just looking at the linux install page and I counted 9 different install methods so I don't know if this covers all the bases, but I guess that's why there's a designated libraries directory separate from the app install.

etjones, 2022-02-15:
Should be fixed in v1.1.3, on PyPI

closed 2022-02-15 by etjones