SolidCode/SolidPython #200: Japanese text became crash when generate scad file

potty26, 2022-11-19:
japanese text became crash when using text(). 
I can't find any solution.
Thank you for your help.

[image]
[image]

jeff-dh, 2022-11-19:
```python
>>> from solid2 import *
>>> t = text("日本語")
>>> scad_render(t)
'\n\n\n\ntext(text = "日本語");\n'
>>> scad_render_to_file(t)
```

seems to work (assuming my input really is japanese text ;).....

OpenScad has some issues with the default font, but it does not crash.

Could you provide more details, an example file that crashes?

On Sat, 19 Nov 2022 02:57:32 -0800
potty ***@***.***> wrote:

> japanese text became crash when using text(). 
> I can't find any solution.
> Thank you for your help.
> 
> [image]
> [image]
> 
>

jeff-dh, 2022-11-19:
buuut, it still seems to work on my system (assuming my input is some japanese text):

```python
>>> from solid import *
>>> t = text("日本語", font="Yu Gothic UI:style=Bold", size=20)
>>> l = linear_extrude(10)(t)
>>> scad_render(t)
'\n\ntext(font = "Yu Gothic UI:style=Bold", size = 20, text = "日本語");'
>>> 
```

Is this probably not a solidpython but a issue concerning your system? Is japanese text working in python on your system otherwise?

potty26, 2022-11-20:
Also execute "scad_render(l)" and can execute correctly.
>>> scad_render(l)
'\n\nlinear_extrude(height = 10) {\n\ttext(font = "Yu Gothic UI:style=Bold", size = 20, text = "日本語");\n}'

But when execute "scad_render_to_file(l,"test.scad")" , generated file was crushed.
 >>> scad_render_to_file(l,"test.scad")
'C:/Users/81905/test.scad'

[image]

jeff-dh, 2022-11-20:
Works fine on my system:

```python
// Generated by SolidPython 1.1.1 on 2022-11-20 09:35:41

linear_extrude(height = 10) {
	text(font = "Yu Gothic UI:style=Bold", size = 20, text = "日本語");
}
/***********************************************
*********      SolidPython code:      **********
************************************************
 
from solid import *

t = text("日本語", font="Yu Gothic UI:style=Bold", size=20)
l = linear_extrude(10)(t)
scad_render_to_file(l)
 
 
************************************************/
```

closed 2022-11-28 by potty26