SolidCode/SolidPython #208: Feature request: any live view of models? (POC project included)

dc740, 2023-11-17:
Hi, I've been looking for a way to develop and see the changes in real time. Does the project have one?

I wrote a proof of concept that can be extended if needed, but it currently a hack I prepared in a couple of nights (it would have been an hour if it wasn't for python watchdog module). It loads the SolidPython code, takes the model out of the .py file, creates a scad file and finally uses openscad to create an STL that gets previewed. It refreshes the view whenever you change the python code (and save it)

https://github.com/dc740/real-time-mesh-viewer

The only good thing is that it works fine, and it only requires SolidPython + OpenSCAD installed. But I only used it in Ubuntu. I was hoping there was something everyone is using and that I just didn't know about.

etjones-erisyon, 2023-11-17:
Nice work, Emilio!  

Have you tried just using OpenSCAD as your viewer? 
The loop I've always used has been to run a SolidPython program to create a `.scad` file, then open that in OpenSCAD and make sure `Design` -> `Automatic Reload and Preview` is checked.  Then any time I run the SolidPython file again, the result appears in OpenSCAD shortly after. Depending on the complexity of the object, this loop may be faster than generating an `.stl` with OpenSCAD and rendering that. My VSCode setup will save and run a program in one key command, cmd-r, which seems like it's probably just as fast as having an external program monitoring my file for saves (cmd-s) and then running some further transforms on them. 

- You also might take a look at [ViewSCAD](https://github.com/nickc92/ViewSCAD) is a project that puts a live 3D render in a Jupyter notebook. That might be useful if it suits your workflow.

- Also worth noting: All new development is happening in the SolidPython 2 repository, [here](https://github.com/jeff-dh/SolidPython)

dc740, 2023-11-17:
THANK YOU!!! I knew I was missing something. I'm not a regular OpenSCAD user, and I wanted to try SolidPython instead. So I was not aware this was possible.

Regarding SolidPython 2. I actually thought the development was stalled or something like that. It'd be good to put a notice on the readme or something. Now that you mention it I see the solidpython2 package published in pip too.

Again, thank you, I will close this ticket then.

closed 2023-11-17 by dc740