SolidCode/SolidPython #211: combine translate and rotate

MaierJuerg, 2024-09-05:
in openscad I use:
```
translate([10,0,5])
rotate([0,90,0])
cylinder(d=50, h=200);

```
can't figure out how to write that with solid as it is not accepting this statement:
`translate((10,0,5))rotate((0,90,0))(cylinder(d=50, h=200))`

jeff-dh, 2024-09-05:
> [Verfolgungsbild][https://github.com/notifications/beacon/AAINZEINKMAOPU43QSDXT3LZVAJOLA5CNFSM6AAAAABNV6VN22WGG33NNVSW45C7OR4XAZNFJFZXG5LFVJRW63LNMVXHIX3JMTHJK37KGU.gif]

cylinder(...).translate(...).rotate(...)

or:

rotate(...)(translate(...)(cylinder(...)))

and / or take a look at the examples....

MaierJuerg, 2024-09-05:
I have installed solid as pip install solidpython
it is version 1.1.3 and I run it with python 3.12 on ubuntu
rotate(...) will add a colon ; after the command making it useless!
runtime error for .rotate
'cylinder' object has no attribute 'rotate'

jeff-dh, 2024-09-05:
https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/05-access-style-syntax.py

https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/04-convenience.py

MaierJuerg, 2024-09-05:
thanks but these examples use solid2. how to install that?

MaierJuerg, 2024-09-05:
Great, thanks, looks much better now!

closed 2024-09-05 by MaierJuerg