SolidCode/SolidPython #212: Any way to sed resolution of a model?

CeSiumUA, 2024-10-05:
Hi!
I'm following the OpenSCAD tutorial, and they use these special variables to increase the model resolution:
```
$fa = 1;
$fs = 0.4;
```
But I can't find a way to do it in a SolidPython, is it possible?

jeff-dh, 2024-10-05:
Take a look at soldpython2 (the maintained fork of solidpython):

https://github.com/jeff-dh/SolidPython/blob/389a575206ca4b647ba3c1a597cfacb32b0488c8/solid2/examples/01-basics.py#L5

CeSiumUA, 2024-10-05:
@jeff-dh  Thanks!

CeSiumUA, 2024-10-05:
For those, who are interested how to do it in the current version of SolidPython:
Just write that code to file header while rendering to file:
`scad_render_to_file(res, 'cube.scad', file_header='$fa = 1;$fs = 0.1;')`

closed 2024-10-05 by CeSiumUA

jeff-dh, 2024-10-05:
At object level try `segments`

```
sphere(1, segments=128]
```

CeSiumUA, 2024-10-08:
@jeff-dh Thanks, that's also useful!