SolidCode/SolidPython #214: scad_tokens.py:55: SyntaxWarning: invalid escape sequence '\&'

andyDoucette, 2025-02-01:
A script to reproduce using [this](https://github.com/BelfrySCAD/BOSL2/commit/5b37739ab1666ad3c0a919bd5993f3854fd16cb7) version of BOSL2.

fill.py
```python
#!/usr/bin/python3

from solid import *

b2=include('BOSL2/std.scad')
```

```sh
[andromodon@spectre fill]$ ./fill.py 
/usr/local/lib/python3.12/dist-packages/solid/py_scadparser/scad_tokens.py:55: SyntaxWarning: invalid escape sequence '\&'
  t_AND = "\&\&"
/usr/local/lib/python3.12/dist-packages/solid/py_scadparser/scad_tokens.py:56: SyntaxWarning: invalid escape sequence '\|'
  t_OR = "\|\|"
```

The fix, in scad_tokens.py:
```
t_AND = r"\&\&"
t_OR = r"\|\|"
```

jeff-dh, 2025-02-03:
Especially if you want to use bosl2, you  might want to have a look at
SolidPython2:

https://github.com/jeff-dh/SolidPython
https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/07-libs-bosl2-logo.py
https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/07-libs-bosl2.py

Even though it might have the same issues (the raw strings).

andyDoucette, 2025-02-03:
Thank you!  I'm a little confused about what's solidpython2 and what's
solidpython.  There are lots of places where the names are used
interchangeably, it seems.   Can you clarify?  Why change the name instead
of just having a version?

jeff-dh, 2025-02-03:
SolidPython2 was an experimental fork which matured and "finally
took over" and we didn't merge it back for whatever reasons.

If you're interested in the discussion:

https://github.com/jeff-dh/SolidPython/issues/2

Sorry for the confusion.